aws/aws-fpga #586: Building DCP fails when using sh_ddr

ChrisKjellqvist, 2022-11-03:
Using the master branch.
If I do not use the sh_ddr IP by setting `DDR_A_PRESENT(0)` for all of the DDR channels, then everything synthesizes without issue.

Whenever I do use the sh_ddr IP, it results in the following synthesis error:
```
INFO: [Synth 8-6157] synthesizing module 'lib_pipe' [/home/centos/aws-fpga/hdk/common/shell_v04261818/design/lib/lib_pipe.sv:27]
INFO: [Synth 8-6155] done synthesizing module 'lib_pipe' (63#1) [/home/centos/aws-fpga/hdk/common/shell_v04261818/design/lib/lib_pipe.sv:27]
ERROR: [Synth 8-5809] Error generated from encrypted envelope. [/home/centos/aws-fpga/hdk/common/shell_v04261818/design/sh_ddr/synth/sh_ddr.sv:606]
ERROR: [Synth 8-5809] Error generated from encrypted envelope. [/home/centos/aws-fpga/hdk/common/shell_v04261818/design/sh_ddr/synth/sh_ddr.sv:580]
ERROR: [Synth 8-6156] failed synthesizing module 'sh_ddr' [/home/centos/aws-fpga/hdk/common/shell_v04261818/design/sh_ddr/synth/sh_ddr.sv:32]
ERROR: [Synth 8-6156] failed synthesizing module 'composer_aws' [/home/centos/Composer/Composer-Hardware/vsim/design/composer_aws.sv:27]
```

Further up in the log is the following which may be related.
```
WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Implementation target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/centos/aws-fpga/hdk/common/shell_v04261818/design/ip/src_register_slice/src_register_slice.xci
/home/centos/aws-fpga/hdk/common/shell_v04261818/design/ip/dest_register_slice/dest_register_slice.xci
/home/centos/aws-fpga/hdk/common/shell_v04261818/design/ip/axi_register_slice_light/axi_register_slice_light.xci
/home/centos/aws-fpga/hdk/common/shell_v04261818/design/ip/axi_register_slice/axi_register_slice.xci
```

Any help with this issue would be appreciated. Otherwise, I could generate another DDR controller IP using Vivado though the documentation seems to indicate that sh_ddr needs to be used in order for the toolchain to work correctly.

AWScsaralay, 2022-11-04:
Hello,

Thanks for reaching out to us on this topic. Could you please help us answer following questions?

* Are you seeing this error when instantiating `sh_ddr.sv` and setting `DDR_A_PRESENT=0`? Or does it error out with `DDR_A_PRESENT=1` too?
* Were you able to successfully build our example [CL_DRAM_DMA](https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/design/cl_dram_dma.sv#L536-L538) design using the desired setting for `sh_ddr.sv`?
* Could you please confirm which tool version are you using?
* Were there any errors upon running `source hdk_setup.sh`? https://github.com/aws/aws-fpga/blob/master/hdk_setup.sh ?

Thank you!
Chakra

closed 2022-11-07 by ChrisKjellqvist

ljp0101, 2023-10-13:
Just wasted time on this same and I second the @ChrisKjellqvist pull request.

The setup script doesn't work, and worse misleadingly implies it should work with DDR A/B/D if some commented code is uncommented (ignoring the obvious error about using read_bd on a .xcix). And nothing about what's required by the encrypted envelope is documented anywhere; we have to guess based off the examples.

czfpga, 2023-10-23:
Hi,

Thank you for the feedback. I'll bring this to the dev team. 

Chen