aws/aws-fpga #592: MSI-X interrupts from CL are not supported when using SHELL without XDMA?

ehudeliaz, 2022-12-12:
Hi,

I am using the small shell and I want to implement MSI-X interrupts from the CL to the SW.

It's not clear to me from the documentation if the interfaces for the CL MSI-X interrupts are usable, when there is no XDMA in the FPGA.
Looking from the code of the example designs, I see that those interrupts requests are set to 0 when the SHELL doesn't have XDMA. 
I also see that the XDMA maps the interrupts to the MSI-X BAR.
Lastly, the MSI-X BAR is not accessible to the CL as master, so I can't implement my own logic to generate MSI-X by writing to the BAR.

Is that correct and MSI-X from the are not supported by **HW** when there is no XDMA?
If not, then how should one implement it, in the CL?

Thanks!

AWScsaralay, 2022-12-21:
Hi, 

MSI-X in F1 is supported by both - XDMA Shell and Small Shell. Small Shell is essentially a type of F1 shell that doesn't contain a DMA engine (XDMA). Small Shell examples below exercise the MSI-X interrupts using CL_DRAM_DMA example design:

https://github.com/aws/aws-fpga/tree/small_shell/hdk/cl/examples/cl_dram_dma#test_dram_dmac
https://github.com/aws/aws-fpga/blob/small_shell/hdk/cl/examples/cl_dram_dma/software/runtime/test_dram_dma.c#L89-L93
https://github.com/aws/aws-fpga/tree/small_shell/hdk/cl/examples/cl_dram_dma#dram-dma-example-metadata

The MSI-X BAR is accessible only from the Host PCIe to configure the MSI-X table. Once the MSI-X table is configured, the CL may trigger an interrupt by pulsing any of the bits in `cl_sh_apppf_irq_req[15:0]` and wait for an ack on corresponding bit `sh_cl_apppf_irq_ack[15:0]`

Please refer to https://github.com/aws/aws-fpga/blob/small_shell/hdk/docs/AWS_Shell_Interface_Specification.md#interrupts on supported interrupt functionality. Let us know if you have any questions. 

Thanks!
Chakra

ehudeliaz, 2022-12-22:
Hi Chakra,

Yes, I already understood it from a different perspective. 
I think that the documentation is a bit unclear in this matter, and I would add the next points to make it clearer:
- There is an interrupts controller hardware which exists in the small shell and in the big shell
- This interrupts controller hardware is configured and controlled by the XDMA driver, not the XDMA hardware
- Even if there is no XDMA hardware in the small shell, you can still use the XDMA driver

From the next part in the documentation I understood that the XDMA hardware must exist for the interrupts to work:
"Mapping registers in the DMA controller map the 16 user interrupt sources to MSI-X vectors"

Which is not correct.

Thanks,
Ehud

closed 2022-12-22 by ehudeliaz

AWScsaralay, 2022-12-22:
Hello Ehud, 

Thank you for the feedback as we will continue to improve documentation based on user experience. We are glad you were able to make progress using small_shell. 

Please feel free to reach out to us if you have any questions. 
Thanks!
Chakra