aws/aws-fpga #593: cl_hello_world_ref_hx fails to synthesis

johnrabo, 2022-12-13:
When running through the cl_hello_world_ref_hlx when I get to the following step
`
Implementing the Design
To run implmentation from within the GUI is opened, in the Design Runs tab:
Right click on impl_1 in the Design Runs tab and select Launch Runs…
Click OK in the Launch Runs Dialog Box.
Click OK in the Missing Synthesis Results Dialog Box
This will run both synthesis and implementation.
`

I'm getting the following error:
NFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2021.2/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'cl_f1_inst_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/cl_f1_inst_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_f1_inst_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/cl_f1_inst_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_f1_inst_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/cl_f1_inst_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_f1_inst_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/cl_f1_inst_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_hello_world_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_hello_world_0_0/cl_hello_world_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_hello_world_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_hello_world_0_0/cl_hello_world_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_hello_world_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_hello_world_0_0/cl_hello_world_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_hello_world_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_hello_world_0_0/cl_hello_world_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_system_ila_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_system_ila_0_0/cl_system_ila_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_system_ila_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_system_ila_0_0/cl_system_ila_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_system_ila_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_system_ila_0_0/cl_system_ila_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'cl_system_ila_0_0' generated file not found '/data/aws-fpga-git/hdk/cl/examples/cl_hello_world_ref_hlx/build/scripts/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_system_ila_0_0/cl_system_ila_0_0_sim_netlist.vhdl'. Please regenerate to continue.
CRITICAL WARNING: [filemgmt 56-176] Module references are not supported in manual compile order mode and will be ignored.
add_files: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 2706.898 ; gain = 0.000 ; free physical = 220470 ; free virtual = 245927
INFO: [Common 17-206] Exiting Vivado at Tue Dec 13 21:37:44 2022...
ERROR: [Common 17-107] Cannot change read-only property 'is_locked'.
Resolution: Please refer to Vivado Properties Reference Guide (UG912) for more information on setting properties.
sourcing script /data/aws-fpga-git/hdk/common/shell_v04261818/hlx/build/scripts/subscripts/synth_design_post.tcl failed
INFO: [Common 17-206] Exiting Vivado at Tue Dec 13 21:37:45 2022...

kyyalama2, 2023-01-11:
Dear customer,

Can you please provide the following information

1.) Are you using a developer AMI provided by AWS or are you using on-premise setup for running the development flow?
2.) Can you try this flow on a new clone of the developer kit without running any other flow on the clone before running the hlx example referred to and see if you are still seeing the issue

Thanks

Gogul-N, 2023-01-25:
Hi,

I also have the same issue. I am unable to proceed further. 

I am trying to create .tar file using vivado. I am trying with IPI flow. Kindly help me solving this. Even I have done clone again and ran. But the same error I am getting. 

**ERROR: [Common 17-107] Cannot change read-only property 'is_locked'.**
Resolution: Please refer to Vivado Properties Reference Guide (UG912) for more information on setting properties.
sourcing script /home/centos/src/project_data/aws-fpga/hdk/common/shell_v04261818/hlx/build/scripts/subscripts/synth_design_post.tcl failed

Thank you.

kyyalama2, 2023-01-25:
Dear customer

Thanks for providing this information. Just to make sure, are you running this flow on one of the dev AMI provided by AWS? or are you using your on-premise setup for the run? What tool version are you using

Thanks

Gogul-N, 2023-01-26:
Thank you @kyyalama2 for the reply.

I am using dev AMI and vivado 2021.2
Even I have created AFI for my own custom logic. I did not test it yet. 
Please help me, how to rectify this problem? 
Or shall I run by commenting the same line for my upcoming works?

Thank you.

Gogul-N, 2023-02-01:
Hi @kyyalama2 
Is there any separate file for understanding all functions to be used in C for testing our FPGA Design..?
If so, Please provide.
Thank you.

kyyalama2, 2023-02-22:
Dear customer

So are you trying to use our developer kit for your end-to-end design? If yes, doing source sdk_setup.sh should setup all the required libraries for basic fpga management etc. Is this what you are asking?

Gogul-N, 2023-02-23:
Thank you @kyyalama2  for the reply.

Yes, I want to validate WLAN. So I am trying to pack this WLAN as an IP. And I am going to connect it with AWS f1 instance using aws::create_ipi 
Am I in a right path? 

And still I am facing this issue https://github.com/aws/aws-fpga/issues/593#issuecomment-1403247074  

Kindly help me @kyyalama2 

Thank you

kyyalama2, 2023-06-01:
closing this ticket since inactive for a while. Also the other referred to ticket is still open. Please feel free to reopen if further help is needed

Thanks

closed 2023-06-01 by kyyalama2