aws/aws-fpga #594: How were the IPs in hdk/common/shell_v04261818/design created without any project

za1nrkhan, 2022-12-29:


za1nrkhan, 2022-12-29:
And I see the files for every IP a collection of the files from the `project.hw`, `project.ip_user_files` and `project.srcs` directories had they been created in a Vivado project

AWScsaralay, 2023-01-11:
Hello, 

The IP and xci files provided in [here](https://github.com/aws/aws-fpga/tree/master/hdk/common/shell_v04261818/design/ip) are not managed by any vivado project (.xpr). These IPs were generated and tested internally by AWS. Only the .xci files and their corresponding output products are released in the above directory. 

However for customer IPs, we strongly recommend to create and manage the IPs using vivado a project.

Please let us know if you have any questions. 

Thanks!
Chakra

closed 2023-05-25 by kyyalama2