aws/aws-fpga #595: Changing A1 frequency

abhaywindsor, 2023-01-03:
Our design is able to close timing at 200Mhz when running Vivado with our own scripts. However, with AWS scripts we can only get to ~150Mhz. As per Xilinx, Vivado seems to give up optimizations due to over-constrained clock at 250Mhz. We tried changing the aws_gen_clk_constraints.tcl to change A1 period to 5ns instead of 4ns, but it does not take affect as per the timing reports. We could not figure out where the aws_gen_clk_constraints.tcl setting is being overridden. Can you please provide some more guidance on that?

czfpga, 2023-01-03:
Hi,

Can you please confirm if the target design is being built using the [HDK flow](https://github.com/aws/aws-fpga/tree/master/hdk)? If it is, please note that arbitrary clock frequency is not supported. Instead, F1 provides a list of support clocks as shown in [clock group recipes](https://github.com/aws/aws-fpga/blob/master/hdk/docs/clock_recipes.csv). You should consider using one of the clocks with a desired frequency from the recipe by following [these steps](https://github.com/aws/aws-fpga/tree/master/hdk#step-2-build-the-cl).

Thanks,

Chen

abhaywindsor, 2023-01-03:
Yes, we are using HDK flow. Based on the clock recipe table, if I want to target 200Mhz, I can tie clk_extra_c0 to CL logic. I can also connect sh_ddr to clk_extra_c0. How do I connect PCIe axi bus to clk_extra_c0? Can the asynchronous clocks be avoided?

czfpga, 2023-01-05:
Hi,

The shell interfaces are all tied to clk_main_a0 and that cannot be modified. So you'll need clock conversion when interfacing with the shell. But in this particular case, there won't be any performance loss from the clock conversion. Is the existing design already size  constrained? 

Thanks,

Chen

closed 2023-02-13 by abhaywindsor