aws/aws-fpga #596: PCIS interface AXI ID

abhaywindsor, 2023-02-13:
The AXI ID is 16 bits wide. How many bits are actually used? It seems only 6 bits.
"The AXI ID can be used to determine the source of the transaction: 0x20 : PCI Interface"
What is the maximum number of outstanding reads/writes can SHELL initiate to CL on this interface?

AWScsaralay, 2023-02-22:
Hello, 

The Shell supports a maximum of 32 outstanding transactions on PCIS interface to the CL:

" The SH supports a maximum of 32 transactions outstanding for each type (read/write)..." in:
https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#dma_pcis-interface-timeout-details

Please reach out to us if you have any questions. 

Thanks!
Chakra

kyyalama2, 2023-05-25:
Dear customer

closing this ticket since no outstanding follow up requests on this issue. Please feel free to reopen if you need further help

Thanks