aws/aws-fpga #597: Hi,

Gogul-N, 2023-02-23:
Hi,

I also have the same issue. I am unable to proceed further. 

I am trying to create .tar file using vivado. I am trying with IPI flow. Kindly help me solving this. Even I have done clone again and ran. But the same error I am getting. 

**ERROR: [Common 17-107] Cannot change read-only property 'is_locked'.**
Resolution: Please refer to Vivado Properties Reference Guide (UG912) for more information on setting properties.
sourcing script /home/centos/src/project_data/aws-fpga/hdk/common/shell_v04261818/hlx/build/scripts/subscripts/synth_design_post.tcl failed

The problem is with the file **make_post_synth_dcp.tcl**

if {${BD_PATH} == "NONE"} {
                #Do Nothing
        } else {
                add_files ${BD_PATH}
                set_property is_locked true [get_files ${BD_PATH}]
        }

I have commented the line **set_property is_locked true [get_files ${BD_PATH}]**

After this it is able to synthesis. But my doubt commenting this line will it create any issues further?

Kindly help me.

Thank you.

_Originally posted by @Gogul-N in https://github.com/aws/aws-fpga/issues/593#issuecomment-1403247074_

ljluestc, 2023-08-13:
The error message you're encountering seems to be related to Vivado's inability to change a read-only property 'is_locked'. Commenting out the line set_property is_locked true [get_files ${BD_PATH}] might have allowed you to proceed with synthesis, but it's important to understand the implications of making such a change.

The is_locked property is used to lock a file, preventing any modifications to it. This property is typically set to prevent accidental changes to important files in your design. By commenting out that line, you are essentially removing the protection that prevents modifications to the specified file.

Gogul-N, 2023-08-14:
Thank you @ljluestc for the reply.

closed 2023-08-18 by Gogul-N