aws/aws-fpga #600: Will there be future support for 2022.x (and 2021.2) tool versions in AL2 AMIs?

abejgonzalez, 2023-03-09:
Based on this table there is support for up to 2021.2 tools in the FPGA Developer AMI's. However, when looking at the AL2 versions of the AMI (https://aws.amazon.com/marketplace/pp/prodview-iehshpgi7hcjg?sr=0-2&ref_=beagle&applicationId=AWSMPContessa) I see that only 2021.1 is supported (AMI version 1.11.3). 

Will future versions of the Xilinx tools be supported in AL2 AMIs like the Centos equivalents? Does this include support for 2021.2?

Thanks.

kyyalama2, 2023-03-09:
Dear customer

Thank you so much for the follow up. Please note that this is not to indicate any discontinued support for AL2 and we currently intend to continue our support for AL2. We are currently prioritizing releases based on requirements and other AMI updates etc. Please feel free to let us know if you have any follow up questions

Thanks

abejgonzalez, 2024-06-30:
**I wanted to follow up and ask if there are any updates to AMIs (in either the Centos/AL2 versions) coming again.** When choosing an AMI through the "Launch an instance" portal, I see that for AMI's there are 3 options:

1. `FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc` in "Community AMIs" - Centos7 based supporting Xilinx 2021.2 tools
2. In "AWS Marketplace AMIs" `FPGA Developer AMI` which when you select defaults to `FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc` as well
3. In "AWS Marketplace AMIs" `FPGA Developer AMI(AL2)` which when you select defaults to `FPGA Developer AMI(AL2) - 1.10.5-62ddb7b2-2f1e-4c38-a111-9093dcb1656f` - AL2 based supporting Xilinx 2020.2 tools.

Since I haven't seen an update to any of the AMIs in over 2 years now, is it reasonable to expect that there will be no other updates in the future? The Centos7-based AMIs are hitting EOL June 30, 2024 (1 day from this post) so they will be out of date w.r.t the OS as-well-as the Xilinx tooling. Additionally, AL2 is now out in favor of AL202* AMIs.

AWSjoeluc, 2025-01-07:
Hello!

Sorry for the delayed response, we recently released the [Ubuntu FPGA Developer AMI](https://aws.amazon.com/marketplace/pp/prodview-f5kjsenkfkz5u) with support for Vivado 2024.1. The AMI is compatible with the latest F1 [XDMA shell](https://github.com/aws/aws-fpga/pull/670) and [small shell](https://github.com/aws/aws-fpga/pull/669) releases in addition to the new [f2 branch](https://github.com/aws/aws-fpga/tree/f2).

closed 2025-04-21 by TrahStevAmz