aws/aws-fpga #602: AWS XVC question

jimmy-adams, 2023-03-19:
Hello,
I noticed that AWS are using the XVC method similar to Xilinx. To test a design with bitstream and iltx file, will that be a must to predesign a bd file containing the same design and ila IP?

kyyalama2, 2023-05-25:
Dear customer

Thank you so much for your interest in F1 instances. We have a detailed guide on using Virtual JTAG with F1 instances. Please refer to https://github.com/aws/aws-fpga/blob/master/hdk/docs/Virtual_JTAG_XVC.md for the detailed instructions. Please let us know if you still have any questions on this. You can use Virtual JTAG with our HDK design flow on F1 instances

Thanks

kyyalama2, 2023-06-01:
Closing this ticket since no outstanding request related to this issue. Please feel free to reopen or create a new ticket if further help is needed

closed 2023-06-01 by kyyalama2