aws/aws-fpga #603: AHB Slave interface issue in Vivado IP (AWS f1 Instance)

Gogul-N, 2023-03-31:
Hi,

I am able to create AHB Slave interface in local machine in vivado IP design.

But **I am unable create AHB Slave interface in AWS Server**. In AWS server AXI Slave interface I am able to create and connect it using pencil tool in vivado ip. I am facing problem with AHB interface.

In local machine it is working perfectly not in aws cloud vivado. I need it in aws server.

Kindly help me.

Thank you.

kyyalama2, 2023-05-25:
Dear customer

Can you provide more information on:
1.) Is your design on the AWS cloud using one of the provided FPGA dev AMIs or are you using your own Vivado setup/license?
2.) If using AWS dev AMI, can you please provide which version of the dev AMI are you using
3.) Can you also confirm if the design is targeted for F1
4.) Is there a specific error you are seeing when trying to add AHB slave interface

Thanks

Gogul-N, 2023-06-02:
Hi @kyyalama2 

1) I am using AWS cloud (FPGA dev AMIs)

2) AMI Version:        1.12.1
    Xilinx Version:     2021.2

3) Yes, design is targeted for F1

4) No error I am getting. Actually after connecting the AHB Master ports individually to my design, I am unable to assign any address. It is showing that incomplete path. I cannot connect as a grouped signal since the AHB slave signals are not grouped in my design. Even I tried to group it, but it is not grouping. Just I tried grouping in my local machine vivado , it is working. But not in aws vivado version 2021.2

Thank you