aws/aws-fpga #604: [Question]: Hold iolations in PCIS interface

sajjadahmed677, 2023-04-04:
Hi all,

I am building my design which uses PCIS interface but there are hold violations reported in the PCIS interface. If anyone could help me to resolve these.

Thanks,

```+--------------------------+--------------------------+----------------------------------------------------------------------------------------------------------+
|       Launch Clock       |      Capture Clock       |                                                 Pin                                                      |
+--------------------------+--------------------------+----------------------------------------------------------------------------------------------------------+
|              clk_main_a0 |              clk_main_a0 |                                   WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/m_valid_i_reg/D|
|              clk_main_a0 |              clk_main_a0 |                                   WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/s_ready_i_reg/D|
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/m_payload_i_reg[203]/D|
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/m_payload_i_reg[514]/D|
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/m_payload_i_reg[239]/D|
| WRAPPER_INST/CL/clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/skid_buffer_reg[203]/D|
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/m_payload_i_reg[290]/D|
| WRAPPER_INST/CL/clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/skid_buffer_reg[290]/D|
| WRAPPER_INST/CL/clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/skid_buffer_reg[371]/D|
| WRAPPER_INST/CL/clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/skid_buffer_reg[239]/D|
| WRAPPER_INST/CL/clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/r.r_pipe/skid_buffer_reg[514]/D|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[403]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[251]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[123]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[402]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[250]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[122]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[411]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[303]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[399]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[315]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[302]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[410]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[398]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[314]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[494]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[495]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[463]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[115]/CE|
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[83]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[467]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[462]/CE|
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[82]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[466]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[114]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[335]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[367]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[351]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[239]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[366]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[350]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[334]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[238]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[425]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[424]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[307]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[306]/CE|
[... cut ...]
|              clk_main_a0 |              clk_main_a0 |                            WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[36]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[324]/CE|
|              clk_main_a0 |              clk_main_a0 |                           WRAPPER_INST/SH_SHIM/DMA_PCIS_AXI_REG_SLC/inst/w.w_pipe/m_payload_i_reg[200]/CE|
+--------------------------+--------------------------+----------------------------------------------------------------------------------------------------------+```

AWSjoeluc, 2023-04-05:
Hello!

We will try our best to help out. First, I have some clarifying questions:

1. Is the CL logic interfacing with PCIS ports running in the same `clk_main_a0` domain?
2. Which [clock recipe](https://github.com/aws/aws-fpga/blob/master/hdk/docs/clock_recipes.csv) are you using in your build?
3. Do you see any other violating paths in the design that are worse than the PCIS paths show above?

sajjadahmed677, 2023-04-05:
Thanks @AWSjoeluc  for the response,

1) yes the logic interfacing with PCIS is in same clock domain ```clk_main_a0```.
2) the clock recipe is A0.
3) i don't see any other violation.

AWSjoeluc, 2023-04-05:
@sajjadahmed677 Can you try building with different strategies?
- https://github.com/aws/aws-fpga/blob/master/hdk/common/shell_v04261818/build/scripts/aws_build_dcp_from_cl.sh#L21

If you can't find a passing strategy, would you mind providing a timing report for the failing path?

Thanks!
-Joe

kyyalama2, 2023-06-01:
Closing this ticket since the ticket is inactive for a while. Please feel free to reopen if further help is needed

Thanks

closed 2023-06-01 by kyyalama2