aws/aws-fpga #606: XDMA no longer builds in Ubuntu AMI

rsnikhil, 2023-04-27:
I am using the following AMI:

        AMI ID:   ami-052efd3df9dad4825
        AMI name: ubuntu/images/hvm-ssd/ubuntu-jammy-22.04-amd64-server-20220609

in which I have successfully built and used the XDMA driver for many months now.

The XDMA driver recently stopped working, and I'm trying to re-build it in a recent clone of:

        https://github.com/aws/aws-fpga

in directory:

        aws-fpga/sdk/linux_kernel_drivers/xdma/

Now, 'make' (to make xdma.ko) no longer works.
The **attached transcript** shows numerous compile-time errors.

Possible reason: it seems that there may have been an "unattended upgrade" of the Linux in my Ubuntu instance, from 5.15 to 5.19, and those pci functions seem to have been actually removed.

[xdma_build_errs.txt](https://github.com/aws/aws-fpga/files/11345977/xdma_build_errs.txt)

kyyalama2, 2023-05-02:
Dear customer,

I also followed the exact same sequence of steps on the AMI you provided.

Makefile:25: XVC_FLAGS: .
make -C /lib/modules/5.15.0-1011-aws/build M=/home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-1011-aws'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
/home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/Makefile:25: XVC_FLAGS: .
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/libxdma.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma_cdev.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/cdev_ctrl.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/cdev_events.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/cdev_sgdma.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/cdev_xvc.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/cdev_bypass.o
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma_mod.o
  LD [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma.o
/home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/Makefile:25: XVC_FLAGS: .
  MODPOST /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/Module.symvers
  CC [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma.mod.o
  LD [M]  /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma.ko
  BTF [M] /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma.ko
Skipping BTF generation for /home/ubuntu/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-1011-aws'
ubuntu@<>:~/aws-fpga/sdk/linux_kernel_drivers/xdma$ ls
10-xdma.rules   cdev_bypass.c  cdev_ctrl.o    cdev_sgdma.h  cdev_xvc.o  libxdma_api.h                      xdma.ko     xdma.o       xdma_install.md
Makefile        cdev_bypass.o  cdev_events.c  cdev_sgdma.o  libxdma.c   modules.order                      xdma.mod    xdma_cdev.c  xdma_mod.c
Module.symvers  cdev_ctrl.c    cdev_events.o  cdev_xvc.c    libxdma.h   user_defined_interrupts_README.md  xdma.mod.c  xdma_cdev.h  xdma_mod.h
README.md       cdev_ctrl.h    cdev_sgdma.c   cdev_xvc.h    libxdma.o   version.h                          xdma.mod.o  xdma_cdev.o  xdma_mod.o

I do see xdma.ko generated

rsnikhil, 2023-05-02:
Thank you for your response.

As I tried to explain in my original post, yes, it DOES work with the
AMI (which contains Linux kernel 5.15) when it is freshly installed.

But once you create an instance, after some time it automatically upgrades the kernel to Linux kernel 5.19.

And then, 'make' (for xdma.ko) breaks, as shown in the transcript I attached.

Please try to update the Linux in your instance to reach kernel version 5.19, and then you will see the error.

kyyalama2, 2023-05-03:
Dear customer

Thanks for the follow up. Just to make sure, is the below sequence of steps is what you are referring to

1.) Create an instance using the AMI ami-052efd3df9dad4825
2.) Now wait for time t, before building xdma driver?
3.) after time t, make on xdma driver does not work and kernel is upgraded automatically to 5.19

if thats what you are referring to, can you please let us know how long do you expect to wait (time t?) before you can no longer build xdma driver

Thanks

rsnikhil, 2023-05-03:
The key point is that an instance using Ubuntu AMI starts with Linux
kernel 5.15, with which one *can* successfully build xdma.ko.

The default Ubuntu installation allows so-called "unattended
upgrades", by which it self-upgrades to newer versions of the kernel
over time (no user intervention).  When it upgrades to Linux kernel
5.19, then xdma.ko stops building, with the compile-time errors I
attached in my original post.  Some of the API calls in xdma's C files
have been removed (or renamed) in 5.19.

I do not know how long ("time t") it will take to self-upgrade.  I
myself have been running an instance with that AMI for over 6 months,
but this upgrade happened some time in late March/early April.

Rather than waiting for some unknown time t for the automatic
self-upgrade, if you are an Ubuntu/Linux expert (or if you can get
help from such an expert), I suggest immediately *forcing* an upgrade
to kernel 5.19, after which you'll be able to see the errors.

Thanks.

kyyalama2, 2023-05-08:
Dear customer

Thank you so much for the detailed information. We updated the Ubuntu kernel to 5.19 and tried to build xdma driver. With this flow, we did encounter some issues and the build did not go through successfully. We followed up with Xilinx on this issue and below is the guidance from them

"Supported OS’s for Vitis/Vivado only go up to UB 22.04/up to 5.17 only.
This is even for 2023.1
So yes , the automatic update on the UB ami’s need to be disabled"

So kernel versions above 5.17 currently are not supported by Xilinx tools. Please let us know if you have any comments/need additional information

Thanks

rsnikhil, 2023-05-08:
Thank you for examining this issue, including the follow-up with
Xilinx, and the explanation.

Ok: we understand now that we will have to wait for Xilinx to upgrade
    their XDMA driver so that it will build under kernel 5.19.

QUESTION: Is there any difference between the XDMA driver provided by AWS, at:

    https://github.com/aws/aws-fpga/
    in directory:    sdk/linux_kernel_drivers/xdma/

and the XDMA driver provided by Xilinx at the link below?

    https://github.com/Xilinx/dma_ip_drivers.git

i.e., has the AWS version been modified from the Xilinx version in any significant way?

Thank you.

kyyalama2, 2023-05-24:
Dear customer

Please note that the top of tree code from Xilinx DMA github is not exactly same as the driver provided as part of AWS github. In addition, from the above information provided by Xilinx, I think the limitation might be applicable to their tot github DMA driver repo too. Are you able to get tot DMA driver from xilinx github functional with kernel 5.19?

Thanks

rsnikhil, 2023-05-26:
Dear @kyyalama2:  I have not tried to build the Xilinx GitHub version of XDMA with linux kernel 5.19, so I don't know if it will work there or not. I agree that I think it is unlikely..

kyyalama2, 2023-06-01:
got it. Yes this version is not officially tested for the F1 worflow. Please let us know if you still have any follow up questions on this issue or if we can close this. Thank you so much for your continued interest in F1.

Thanks

closed 2023-08-17 by kyyalama2