aws/aws-fpga #607: Difficulty creating fpga image using HDK getting started guide (INACCESSIBLE_INPUT: Error accessing resource in S3)

hubregtsen, 2023-05-17:
I am following the AWS FPGA Hardware Development Kit (HDK) getting started guide.

I was able to get my `FpgaImageId` when creating using `aws ec2 create-fpga-image`. However, when I use `describe-fpga-images` to see the status, I get
`{
    "FpgaImages": [
        {
            "UpdateTime": "2023-05-17T15:54:55.000Z",
            "Name": "fpga-name",
            "Tags": [],
            "DataRetentionSupport": false,
            "FpgaImageGlobalId": "agfi-[removed]",
            "Public": false,
            "State": {
                "Message": "INACCESSIBLE_INPUT: Error accessing resource in S3. Check permissions, naming, and ensure the bucket is within the same region as the API endpoint.",
                "Code": "failed"
            },
            "OwnerId": "836978898731",
            "FpgaImageId": "afi-[removed]",
            "CreateTime": "2023-05-17T15:54:55.000Z",
            "Description": "fpga-desc"
        }
    ]
}`

I create using
' aws ec2 create-fpga-image         
          --region us-west-2
          --name fpga-name
          --description fpga-desc
          --input-storage-location Bucket=myfolder/tarball,Key=23_05_17-102317_Developer_CL.tar
          --logs-storage-location Bucket=myfolder/logs,Key=logs'
where myfolder is an obscured name. 

When logging in to the AWS Bucket, I see my folder, the tarbal and logs folder, and the .tar and logs file. 

I suspect permission settings being wrong. At this point I have set both the folder rights and object rights to public, yet I still get the error. 

Any tips on what could solve it, or how I could investigate it? Any additional commands you'd like me to run to provide more information?

kyyalama2, 2023-05-25:
Dear customer

One quick check: I see that you are using Bucket=myfolder/tarball, however the actual tarball reference goes under Key=<>.tar as indicated in your message. So the Bucket should just point to the directory that has the tarball and not include the tar ball name. Can you please try your flow with this

Thanks

kyyalama2, 2023-06-01:
Dear customer

Just checking to see if you still need help on this issue. Please feel free to let us know if you need further help or have any follow up questions

Thanks

closed 2023-08-17 by kyyalama2