aws/aws-fpga #611: fpga_dma_open_queue() failure

ljp0101, 2023-06-02:
> fpga_dma_open_queue() failed; AWS error code: -22; AWS error message: This usually indicates a bug in the fpga image tools, but can also be a symptom of a bug in the code which is using the library in cases where the customer is using the library directly.

ljp0101, 2023-06-02:
I started seeing this error when I pick up spot instances ~3 days ago 5-10% of the time and am trying to understand the cause and how to debug. Think it's happened 5x. Rest of the time it just works. We run in Kubernetes and the F1 images (one Ubuntu for our executable, one AL2 AMI that has Xilinx tools on it) hadn't changed in weeks, although other parts of the cluster might have changed but don't think it should impact this. Throwing the Ubuntu container and restarting doesn't resolve, not sure about the same for AL2 container with XRT. Not a DevOps expert and it's very possible I'm missing some nuance. Any idea what might cause this? Hoping I can manage for now by automating detecting the error and throwing the instance to pick up a new one if it's not obvious what I'm doing wrong.

Error:

> fpga_dma_open_queue()` failed; AWS error code: -22; AWS error message: This usually indicates a bug in the fpga image tools, but can also be a symptom of a bug in the code which is using the library in cases where the customer is using the library directly.

Relevant code:

>    // get dma handle, retry 100x with 10ms gaps on fail
>    for (uint i = 0; i < 100; i++) {
>       this->handle_dma = fpga_dma_open_queue(FPGA_DMA_XDMA, 0, 0, false);
>       if (this->handle_dma >= 0) {
>          break;
>       }
>       if (i == 99) {
>          b_error = true;
>          error_message << "ERROR_BAD_INSTANCE fpga_dma_open_queue() failed; ";
>          error_message << "AWS error code: " << this->handle_dma << "; ";
>          error_message << "AWS error message: " << fpga_mgmt_strerror_long(this->handle_dma);
>          return;
>       }
>       usleep(10000);
>    }

Problematic hosts:

> ip-11-63-15-7.ec2.internal
> ip-11-63-12-21.ec2.internal

AWScsaralay, 2023-10-25:
Hi,

We regret any inconvenience caused to you due to this. Could you please post the FPGA metrics to detect if there were any timeouts? 

https://github.com/aws/aws-fpga/blob/master/hdk/docs/HOWTO_detect_shell_timeout.md#how-to-detect-a-shell-timeout-has-occurred

Also, instead of terminating and starting a new instance, could you please try clearing the FPGA Image using [fpga-clear-local-image](https://github.com/aws/aws-fpga/tree/master/sdk/userspace/fpga_mgmt_tools#amazon-fpga-image-afi-management-tools), followed by reloading your AFI with a `-F` option? That should shorten the downtime in-order to get your application up and running quickly. 

Please let us know if you have any questions. 

Thanks!
Chakra

ljp0101, 2023-10-29:
Checked my logs and this hasn't happened in at least a month so the problem appears to be resolved.

I always run fpga-clear-local-image before loading the AFI so I was following your suggestion when I tried to recover by throwing the container, which didn't work.

closed 2023-10-29 by ljp0101