aws/aws-fpga #613: [Question]: Is it possible to instantiate my own DDR IP or I must use SH_DDR?

ehudeliaz, 2023-06-19:
Hi,

I want to instantiate my own version of DDR4 IP instead of DDR A/B/D controllers that are in the sh_ddr.sv.
From tests I've conducted, it will save me a lot of resources.

I wish to:
- Set DDR_A/B/D_PRESENT to 0 in sh_ddr.sv
- Instantiate my own wrapper of the DDR controllers
- Set the pin assignments and timing constraints   

From the documentation it is not clear to me if it is even possible, or I must use SH_DDR for DDRs A/B/D and as well?

Thanks!

AWSjoeluc, 2023-06-19:
Hi!

Unfortunately, it's not possible to instantiate a custom wrapper of the DDR controller; SH_DDR will always bind the pins and prevent customer pin assignments. Additionally, AWS prevents all customer instantiations of the DDR controllers and encrypts SH_DDR to protect customer data from malicious actors. We understand that the SH_DDR may take up more resources than a custom implementation.

Please follow up if we can help with anything else!
-Joe

kyyalama2, 2023-08-17:
closing the thread as clarification was provided previosuly and no recent activity

closed 2023-08-17 by kyyalama2