aws/aws-fpga #614: [Question]: Xilinx library elaboration error with VCS simulation

sajjadahmed677, 2023-06-19:
Hi, 

I am trying to simulate my design (SHELL + CL) with VCS and getting the following error.

```
Error-[SIMERR_NYE] Object not yet elaborated
/opt/Xilinx/Vivado/2021.1/data/ip/xilinx/fir_compiler_v5_2/hdl/fir_compiler_v5_2_vh_rfs.vhd, 47913
  `protect end_protected
  The referenced object has not yet been elaborated.
```
Can somebody please help me to understand why i am seeing such error or what does this error means. I don't see any direct link of this file (```fir_compiler_v5_2_vh_rfs.vhd```) to my design.

Thanks,

AWSjoeluc, 2023-06-19:
Hi, thanks for reaching out on this issue!

We usually see similar issues when the simulation libraries are not generated and/or compiled correctly.  It appears your design is using a Xilinx IP that requires missing modules in `fir_compiler_v5_2_vh_rfs.vhd`. `fir_compiler_v5_2_vh_rfs.vhd` is one of Xilinx's IP files that must be generated and compiled as part of the Makefile flow when running a simulation. I have a few questions that may help solve the issue:

1. Can you confirm that the VCS simulation libraries were compiled correctly? The simulation libraries are automatically compiled in the first simulation you run with one of our examples. The log should contain the results of that compilation.

2. Are you able to successfully simulate one of our CL examples? If yes, please continue down this list. If not, please try recompiling the simulation libraries, a fresh clone of the repo, or reverting changes that may have affected the Makefile flow.

3. Does the your Makefile include the requirements included in a [CL example Makefile](https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/verif/scripts/Makefile), [Makefile.common.inc](https://github.com/aws/aws-fpga/blob/863d963308231d0789a48f8840ceb1141368b34a/hdk/common/verif/tb/scripts/Makefile.common.inc#L1), and [CL example Makefile.vcs](https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/verif/scripts/Makefile.vcs)?

4. Can you confirm you are compiling [top.vcs.f](https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/verif/scripts/top.vcs.f) for your example? Each of our CL examples has its own `top.vcs.f`. There should be records in the logs indicating that `top.vcs.f` was used to compile the source files and Xilinx IP libraries.

5. Can you confirm you are compiling [tb.vcs.f](https://github.com/aws/aws-fpga/blob/863d963308231d0789a48f8840ceb1141368b34a/hdk/common/verif/tb/filelists/tb.vcs.f)?

Please let us know what you find!
-Joe

kyyalama2, 2023-08-17:
closing the thread since no recent activity/follow up. Please feel free to reopen if further help is needed

closed 2023-08-17 by kyyalama2