aws/aws-fpga #615: Cannot simulate with the branch small_shell

vsoBen, 2023-06-21:
fpga.sv is using module sh_bfm with inputs cl_sh_ddr_awuser and cl_sh_ddr_aruser.

These inputs shall be added in cl_ports_sh_bfm.vh

diff --git a/hdk/common/shell_v04182104/hlx/verif/cl_ports_sh_bfm.vh b/hdk/common/shell_v04182104/hlx/verif/cl_ports_sh_bfm.vh
index a3ab4b8..9c71d75 100755
--- a/hdk/common/shell_v04182104/hlx/verif/cl_ports_sh_bfm.vh
+++ b/hdk/common/shell_v04182104/hlx/verif/cl_ports_sh_bfm.vh
@@ -190,6 +190,9 @@
    input logic [31:0] ddr_sh_stat_rdata1,
    input logic [7:0] ddr_sh_stat_int1,
 
+   input        cl_sh_ddr_awuser,
+   input        cl_sh_ddr_aruser,
+
    output logic [7:0] sh_ddr_stat_addr2,
    output logic sh_ddr_stat_wr2, 
    output logic sh_ddr_stat_rd2,

AWSjoeluc, 2023-06-26:
Hello,

Thank you for opening an issue and proposing these changes! You can also open up a pull request within GitHub to introduce these changes. We'd greatly appreciate more information on how to reproduce this simulation failure.

1. Do you experience a simulation failure due to the missing ports (`cl_sh_ddr_awuser` & `cl_sh_ddr_aruser`) in `cl_ports_sh_bfm.vh`?
    * These ports are non-functional in all of our devkit examples.
2. Do you observe a simulation failure when using one of our devkit examples?
    * If you observed the issue on an AWS example, can you provide the steps you took to setup and run the simulation?
    * If you observed the issue on a custom example, are you able to produce the same issue when using one of the AWS examples?

Thanks again!
-Joe

kyyalama2, 2023-08-17:
closing the thread since no recent activity. Please feel free to reopen if further help is needed. Thanks