aws/aws-fpga #617: Reading Interrupt Signal from the shell

Gogul-N, 2023-06-27:
Hi,

I want to read the signal **irq_req[15:0]** from the shell.

How to read it..?

The document provided in the below link, but I am not able to understand it.
https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#interrupts 

Kindly help me on how to use the MSI-X.

Thank you

AWScsaralay, 2023-07-06:
Hello, 

The `cl_sh_apppf_irq_req[15:0]` should be pulsed by the CL to trigger an MSIX interrupt and wait for a corresponding pulse ack on `sh_cl_apppf_irq_ack[15:0]`. Please note that CL should not send out another interrupt request while still waiting on the ack from the Shell. In your IPI design, you could use the `sh_cl_apppf_irq_ack[15:0]` signal to track any pending interrupts.
https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#interrupts

An example logic for triggering and acking interrupt is shown in the link below: 
https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/design/cl_int_tst.sv#L106-L134

Software test to run MSIX interrupts can be found here.  
https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/software/runtime/test_dram_dma.c#L170-L249

Unfortunately, the Shell does not relay the status of `cl_sh_apppf_irq_req[15:0]` to the host (or users) directly but you could connect this to any of the registers in your design and check the status. Alternately you could also use the `cl_sh_status_vled[15:0]` and use `fpga-get-virtual-led` [Management tool](https://github.com/aws/aws-fpga/tree/master/sdk/userspace/fpga_mgmt_tools) to check the status of interrupts. Please note that these signals are pulsed so you may have to pair it with `sh_cl_apppf_irq_ack[15:0]` to track any pending status. 

Please reach out to us if you have any questions. 

Thanks!
Chakra

Gogul-N, 2023-07-10:
Thank you @AWScsaralay for the reply

closed 2023-07-10 by Gogul-N