aws/aws-fpga #618: vitis_setup.sh with non-root permissions

ajderr2000, 2023-06-28:
We are attempting to run the vitis_setup.sh script in a managed environment which already has all of the pre-reqs (Vivado, Vitis, python, etc) installed and available, but we don't have root access to the host. 

Is there an alternative way to source this vitis_setup.sh script without the requirement for root level access?

linux-guy-217, 2023-07-06:
Yes, the packages required by the script are satisfied and the script has been sourced by root once already on this system. 
When a non-root user tries to source the script on the same host, they are asked for sudo permissions right away. 

When running without root/sudo permissions:

`$ source vitis_setup.sh
INFO: Setting up environment variables
DEBUG: AWS_FPGA_REPO_DIR not set so setting to /path/AWS_VITIS_EX_1/aws-fpga
INFO: Sourcing sdk_setup.sh
INFO: Setting up environment variables
DEBUG: AWS_FPGA_REPO_DIR=/path/AWS_VITIS_EX_1/aws-fpga

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for USERNAME:
-bash: /tmp/sdk_root_env.exp: Permission denied
-bash: /tmp/sdk_root_env.exp: Permission denied
-bash: /tmp/sdk_root_env.exp: Permission denied
-bash: /tmp/sdk_root_env.exp: Permission denied
-bash: /tmp/sdk_root_env.exp: Permission denied

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for USERNAME:

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for USERNAME:
Entering /path/AWS_VITIS_EX_1/aws-fpga/sdk/userspace/utils
rm -f *.o *.a ../lib/libutils.a
rm: cannot remove ‘../lib/libutils.a’: Permission denied
make: *** [clean] Error 1
make clean failed
Error: mkall_fpga_mgmt_tools.sh returned 1
Error: AWS SDK install was unsuccessful, sdk_install.sh returned 0`

kyyalama2, 2023-07-10:
Dear customer

You might be seeing those errors because the script tries to install the required packages which require root permissions by default. You need to manually disable those install steps that are trying to install the packages required. However please make sure you have the exact packages with the exact versions the script tries to install or else might see other issues with further development steps

Please feel free to let us know if you have any follow up questions

Thanks

linux-guy-217, 2023-07-12:
It looks like the main issue is actually in the sdk_setup.sh that is called by the vitis_setup.sh file. 

Lines 33-40 are setting explicit root permissions on a file in /tmp, is this required to be owned and created by the root user? Does the sdk_root_env.exp file need to be writable by any user that is going to be running this vitis_setup.sh? 

```
 33 sudo rm -f /tmp/sdk_root_env.exp
 34 typeset -f allow_non_root > /tmp/sdk_root_env.exp
 35 echo "export AWS_FPGA_SDK_GROUP=${AWS_FPGA_SDK_GROUP}" >> /tmp/sdk_root_env.exp
 36 echo "export AWS_FPGA_SDK_OTHERS=${AWS_FPGA_SDK_OTHERS}" >> /tmp/sdk_root_env.exp
 37 echo "export SDK_NON_ROOT_USER=${SDK_NON_ROOT_USER}" >> /tmp/sdk_root_env.exp
 38 echo "export AWS_FPGA_SDK_OVERRIDE_GROUP=${AWS_FPGA_SDK_OVERRIDE_GROUP}" >> /tmp/sdk_root_env.exp
 39 sudo chown root:root /tmp/sdk_root_env.exp
 40 sudo chmod 700 /tmp/sdk_root_env.exp
```

linux-guy-217, 2023-07-12:
Running as non-root:
Note the error about the missing linux64 file/dir, which is not an error when running as root. 

```
source vitis_setup.sh
INFO: Setting up environment variables
DEBUG: AWS_FPGA_REPO_DIR not set so setting to /path/AWS_VITIS_EX_1/aws-fpga
INFO: Sourcing sdk_setup.sh
INFO: Setting up environment variables
DEBUG: AWS_FPGA_REPO_DIR=/path/AWS_VITIS_EX_1/aws-fpga
Entering /path/AWS_VITIS_EX_1/aws-fpga/sdk/userspace/utils
rm -f *.o *.a ../lib/libutils.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.24"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include  linux64 -c -o io.o io.c
cc: error: linux64: No such file or directory
make: *** [io.o] Error 1
make failed
Error: mkall_fpga_mgmt_tools.sh returned 1
Error: AWS SDK install was unsuccessful, sdk_install.sh returned 0
```

Running as root:
No error about a missing linux64 dir. 

```
source vitis_setup.sh
INFO: Setting up environment variables
INFO: Sourcing sdk_setup.sh
INFO: Setting up environment variables
Entering /path/AWS_VITIS_EX_1/aws-fpga/sdk/userspace/utils
rm -f *.o *.a ../lib/libutils.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.24"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o io.o io.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.24"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o log.o log.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.24"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o logger-kmsg.o logger-kmsg.c
mkdir -p ../lib
ar rcs ../lib/libutils.a io.o log.o logger-kmsg.o
...
INFO: Internet Access OK
INFO: Checking ICD is installed
INFO: Found 'libxilinxopencl.so
INFO: AWS Platform: 201920_3 Vitis Platform is up-to-date
ERROR: Xilinx XRT runtime not installed - This is required if you are running on an F1 instance.
INFO: The default AWS Platform has been set to: "AWS_PLATFORM=$AWS_PLATFORM_201920_3"
INFO: Vitis Setup PASSED
INFO: To run a runtime example, start the MPD service by calling: `systemctl is-active --quiet mpd || sudo systemctl start mpd`
```