aws/aws-fpga #619: AWS XVC new update

jimmy-adams, 2023-07-05:
In last [issue](https://github.com/aws/aws-fpga/issues/602), i posted a question related to the remote debug of AWS.
Thanks alot for the mantainer who gave me some docs for reference.
I found the debug_bridge ip stays with the design in the example, such as the cl_hello_world.sv. However, i am still a little bit confused about the deployment such as the linux with the petalinux. Is there any other configuration needed to enable the debug_bridge function and remote debug?
Thanks in advance

jimmy-adams, 2023-07-11:
> Dear customer
>
> To make sure, are you referring to if there is any setup/configuration required on the host or remote machine that you would like to use for remote debug? is there any specific setup that you are exploring?
>
> Thanks

Hello,
I am curious about the use of debug_bridge in the cl_hello_world example. I am not very clear of it's function.
To be more specific, does the debug_bridge support the virtual or remote programming? because based on the example, i thought the debug_bridge will be generated into the bitstream.

Best Regards

kyyalama2, 2023-07-11:
Dear customer

This debug_bridge is only to support remote debug using Virtual JTAG. This does not support JTAG based programming virtually/remotely. The virtual JTAG flow can be used to monitor any ILA signals using ltx file. The only way to program our FPGAs is to use the AFIs. 

Thanks

kyyalama2, 2023-08-17:
closing the thread since inactive for a while and provided some info Virtual JTAG. Please feel free to reopen if further help is needed. Thanks

closed 2023-08-17 by kyyalama2