aws/aws-fpga #620: ERROR: [Place 30-678] Failed to do clock region partitioning

Gogul-N, 2023-07-10:
Hi,

Getting the following error while implementing my custom logic. 

**Note: I am using Vivado GUI IP Integrator. AMI Version: 1.12.1  Xilinx Version:     2021.2**

ERROR: [Place 30-678] Failed to do clock region partitioning: Cannot find an available clock routing track for clock net WRAPPER_INST/CL/cl_inst/xyz/BUFGCE_inst_0 in its partition defined by a rectangle from clock region X2Y11 to clock region X2Y13. A clock partition is a rectangular area covering all clock loads and the clock region for its clock root. It may cover the clock source as well. Each clock net needs to use the same routing track across all clock regions of its partition. In this case, other clock nets are already using resources in one or more clock regions of this partition.

Overall utilization is very less.

In few runs it is not showing this error and implementation completed successfully.

But in few runs, even without changing the design, the above error is thrown. Kindly help me solving this.

Thank you.

kyyalama2, 2023-07-10:
Dear customer

Do you see this error only for this version of Vivado, or do you also see this issue with other tool versions? Also can you confirm if the clock net referred to is part of your CL design and the constraint was added by your CL?

Thanks

Gogul-N, 2023-07-11:
Thank you @kyyalama2 for the reply

I am using AWS cloud (FPGA dev AMIs) and available version of the vivado is 2021.2.

The clock net referred to is a part of my CL.

And for each new run I am getting error in different clock nets.

Thank you

kyyalama2, 2023-07-11:
Dear customer

Thanks for the inputs. Sounds like this likely is a Vivado tool issue than specific to F1 dev environment. We will point AMD to your request for help. Also please note that since the place and route goes through a variety of optimizations, the results (in terms of which net failed constraints etc) might be varying between runs.

Thanks

Gogul-N, 2023-07-11:
Thankyou @kyyalama2 for the reply.

And waiting for your further help.

kyyalama2, 2023-07-11:
Dear customer

First thing to try is using place_design with the -no_bufg_opt switch as placer adds BUFG which can cause clocking contention.  This usually clears things up.
 
Can you also confirm what added clocking is added to the CL as it seems like added BUFGs are added into the CL?

Thanks

Gogul-N, 2023-07-12:
**Can you also confirm what added clocking is added to the CL?**
     Yes, It is added in the CL
     
Actually in our design we had clock gating (used OR gate for clock gating). While implementing in the clock path LUT was there because of OR gate and it led to more hold violations. Even we enabled clock gating conversion during synthesis but no improvement. So we removed all the OR gates and replaced it with BUFGCE and implemented. With this change timing was met without any issues. But now even without any modifications, this issue I am getting.

Thank you @kyyalama2

Gogul-N, 2023-07-14:
Hi @kyyalama2 
 
I tried placing all BUFGCEs manually as it is recommended here [](https://support.xilinx.com/s/article/67674?language=en_US) 

I did not use any LOC constraint for the BUFGCE previously. Any how according to the second suggestion in the above link, I placed all the BUFGCEs manually using CLOCK_REGION constraint.

After this modification. The error is gone.

Thank you.

kyyalama2, 2023-08-17:
Looks like this issue is resolved. So closing the thread. Thanks

closed 2023-08-17 by kyyalama2