aws/aws-fpga #621: Limitations in the Pblock of Custom Logic

Gogul-N, 2023-07-12:
Hi,

Eventhough the clock region assigned for the custom logic is {CLOCKREGION_X0Y10:CLOCKREGION_X5Y14 CLOCKREGION_X0Y0:CLOCKREGION_X2Y9}, the Vivado tool is always placing my design in CLOCKREGION_X0Y0:CLOCKREGION_X2Y9.

Due to this, congestion is more and I am getting placement issues.
Any way to let the Vivado tool to make use of the complete pBlock  to avoid routing issues?

Kindly help me on this.

Thank you.

czfpga, 2023-07-12:
Hi Gogual-N,

I need your help to clarify the target region you want to use. You mentioned both {CLOCKREGION_X0Y10:CLOCKREGION_X5Y14} and {CLOCKREGION_X0Y0:CLOCKREGION_X2Y9} regions above. Do you have have both constraints defined in the `cl_pnr_user.xdc` file? 

In addition, I would recommend you to review your build log and looking for any warnings/critical warning on the pblock definition constraints. Normally if there is any syntax error in the user constraint file, the constraint will be ignored by Vivado. I Hope this helps.

Thanks,

Chen

Gogul-N, 2023-07-13:
Hi @czfpga 
Thank you for the reply.
I mean, we can use the {CLOCKREGION_X0Y10:CLOCKREGION_X5Y14 CLOCKREGION_X0Y0:CLOCKREGION_X2Y9} as it is given for CL and the remaining region is given for Shell. 

But vivado tool is placing the design only in CLOCKREGION_X0Y0:CLOCKREGION_X2Y9 only and it is not using CLOCKREGION_X0Y10:CLOCKREGION_X5Y14 this region. Due to this restriction, congestion is more and facing issues while routing. 

How to resolve this?

Thank you

czfpga, 2023-07-13:
Hi Gogual-N,

Thank you for the clarification. This constraints only define the overall clock regions of the FPGA that's available to the CL. To help guide Vivado's placement to use the full CL region, you'll still need to use the `add_cells_to_pblock` constraint to assign individual CL module to the region you want. You can refer to one of CL example's place-and-route constraint for all the commands you need. For example, here is a good one: https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_dram_dma/build/constraints/cl_pnr_user.xdc

In addition, please note that in order to make sure the final design can still meet the timing while the CL modules are moved away from each, it's recommended to add pipeline stages whenever necessary to the buses that are used to chain those modules up. I hope this helps.

Thanks,

Chen

Gogul-N, 2023-07-14:
Hi @czfpga (Chen)

Thank you for the reply.

I will try this.

kyyalama2, 2023-08-17:
closing the thread since been inactive for a while. Please feel free to reopen if required. Thanks

closed 2023-08-17 by kyyalama2