aws/aws-fpga #622: Using ARM processor in f1 FPGA instance

Gogul-N, 2023-08-24:
Hi,

I have to connect my design with ARM processor and I have to create 2 AFIs with the same design. One is transmitter and another is receiver. But both have to be configured in a different way so that it will act as transmitter and receiver. But the thing is, register addresses of both transmitter and receiver are same since both AFIs have same offset address. Also, I want to do P2P transmission. How can I achieve this by configuring both registers with same addresses? Also is it possible to connect ARM to AWS Shell? And how to do P2P transmission? The example provided in [github is](https://github.com/awslabs/aws-fpga-app-notes/tree/master/Using-PCIe-Peer2Peer) little confusing.

Thank you.

Gogul-N, 2023-08-28:
Hi @czfpga (Chen),

I am working with f1.4xlarge instance only. I am confused here, consider that same AFI is loaded in both slots. And the register I want to configure is 0x30041001. Slot 0 has to be configured with the value 0x11001001 and the slot 1 has to be configured with the value 0x11000001. How can I achieve this?

Thank you.

Gogul-N, 2023-08-29:
Hi @czfpga (Chen),

Thank you for the reply. I will explore it regarding P2P as your description. 

Actually I have completed receiver validation using f1 instance. Now I have to connect the ARM processor to it. It is a softcore  logic only. My doubt is, which BAR of Shell interface has to be connected to ARM processor soft logic? or with out shell interface can I talk to my CL through ARM or I have to talk to my CL through Shell to ARM to CL?

Thank you

czfpga, 2023-08-29:
Hi @Gogul-N,

Interfacing with the soft core in CL have to be done through the shell. Depending on the type of interface, if the number of signals is small and they're very simple, like DIP switches, you can consider using the [miscellaneous signals](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#misc).  If you need access to the core's control & status registers, you might consider using the [OCL or BAR1 interface](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#axi-lite-interfaces-for-register-access----sda-ocl-bar1) (AppPF, BAR0 or BAR1). 

Thanks,

Chen

Gogul-N, 2023-08-30:
Hi @czfpga (Chen)

Thank you so much for the quick reply. I will continue as per your suggestion.

Thank you.

Gogul-N, 2023-08-30:
Hi @czfpga (Chen)

[image]

This is my exact requirement. The signals A and B from Transmitter should be connected to A and B of the receiver. How can I achieve this? Also both transmitter and receiver will have ARM soft core connected to it.

Thank you.

czfpga, 2023-08-30:
Hi @Gogul-N 

Thank you for clarifying the application. As I mentioned before, F1 does **NOT** expose any physical I/O pins to CL designs. 

So one option here is, if it's applicable, you can consider encapsulating data patterns on signal A/B to AXI-MM transactions and forwarding the transactions to the [PCIM interface](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#pcim-interface----axi-4-for-outbound-pcie-transactions-cl-is-master-shell-is-slave-512-bit) on the Tx instance. Rx will receive them from the [PCIS interface ](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#pcis_interface) and de-capsulate them before sending them to the receiver. However, this option might not work if A/B signals are expected to stream data all the time. F1 doesn't support any direct signal connection between slots.

Thanks,

Chen

Gogul-N, 2023-08-31:
Hi @czfpga (Chen),

Shall I use AXI GPIO for transmitting the signals A, B as well as to receive it..? Actually, there will be continues data transmission. I mean, each clock cycle data will be changing. 

Thank you.

czfpga, 2023-09-01:
Hi @Gogul-N,

That will not work as you expect. All signals are handled by the Shell. And for P2P eventually all data will be encapsulated to PCIe packets in order to be transmitted to the P2P partner. There is NO direct GPIO connection between 2 slots. 

Thanks,

Chen

Gogul-N, 2023-09-01:
Hi @czfpga (Chen) 

Thank you so much for your patience. 

I have understood. 

To communicate with the shell any how we need an anxiety interface from CL then how the signals A and B are connected to Shell without any interface to PCIM/PCIS. Here only I am confused a lot. 

Thank you.

AWScsaralay, 2023-10-25:
Hi, 

As Chen stated previously, there cannot be any direct connections between `A/B` ports of `TX` to `A/B` of `RX` blocks. One possible way you could do is as shown below: 
[image]

However, transferring only 10-bits at a time between two FPGAs will be extremely slow due to PCIe overhead. If your goal is to synthesize a TX and RX block in the same FPGA then it might be simpler to just use a GPIO block or simple signaling/handshake between TX and RX blocks. 

Please let us know if you have any questions. 

Thanks!
Chakra

Gogul-N, 2023-10-26:
Hi @AWScsaralay ,

Thank you so much for your deep explanation. 
I try to synthesize both blocks in the same FPGA if there is enough space. If not, I will go with 2 FPGAs.

Thank you.

Gogul-N, 2023-12-08:
Hi @AWScsaralay ,

Thank you for your reply. I will try and update here.

Thank you.

Gogul-N, 2023-12-11:
Hi @AWScsaralay ,

Thank you for your response. 
If I face any issues while implementing the above steps, I will update it here,

Thank you

closed 2024-03-12 by AWScsaralay