aws/aws-fpga #623: DDR performance drop with specific address access pattern

indragade, 2023-09-27:
Hi, we are seeing performance drop when we access specific address pattern from DDR. Please find the below details.

in a single test we read same address repeatedly as mentioned below

rd_addr is 64B aligned and each read transaction is rd_len=1 (128B access)

'P' is the expected performance number

rd_addr =

0 -> no drop (P)

2 -> no drop (P)

4 -> drop (~P/4)

6 -> drop (~P/4)

8 -> no drop (P)

10 -> no drop (P)

12 -> drop (~P/4)

14 -> drop (~P/4)

..........this pattern repeats........

following are the results if we read more than 1 address repeatedly

{0, 2} -> no drop (P)

{4, 6} -> drop (~P/2)

{2, 4} -> drop (~P/2)

{0, 2, 4, 6} -> no drop (P)

{8 ,10} -> no drop (P)

{4 , 12} -> drop (~P/2)

{4, 6, 8, 10} -> no drop (P)

{4 ,6 ,12, 14} -> no drop (P)

is there any known DDR performance limitation for the above address pattern?

Thanks

Indra

AWScsaralay, 2023-10-25:
duplicate of https://repost.aws/questions/QUc--sBOHYSUmBMtIlBW8lBg/ddr-usage-in-f1-fpga-instance

indragade, 2023-10-26:
Thanks Chakra for the reply.
I guess you might have tagged the wrong repost link.
https://repost.aws/questions/QUPA0GiM6GTBiVzj45Aj0IQg/ddr-performance-drop-with-specific-address-access-pattern

In our case the address access pattern is random and access lengths are short (64B or 128B) 
I assume these are not user configurable at this point to take performance advantage for the access patterns listed above.
         "**ROW_COLUMN_BANK_INTLV**" addressing mode and 
          "**Force Read and Write commands to use AutoPrecharge**"

AWScsaralay, 2023-10-26:
Thank you for the pointer, somehow I copied wrong link from re:post.

Correct, our DDR controllers are not user configurable so random access patterns shall result in poor performance.

indragade, 2023-10-26:
Thanks for confirming !!

closed 2024-03-12 by AWScsaralay