aws/aws-fpga #624: PCIM AXI W Channel hangs with wReady stuck at 0

tewujianke, 2023-09-27:
I observed hanging in FPGA when CL continuously wrote to the host memory through PCIM. Each write request contains 512B (awlen=7). The hanging is only observed at 136MHz CL clk, but not observed when changed to a different clk frequency. When it hangs, the **wReady** signal from the shell seems stuck at 0. The image has to be reloaded to resolve the hanging.

fpga-describe-local-image -S 0 --metrics returns all 0 in error/timeout counters. Any advice?

```
fpga-describe-local-image -S 0 --metrics
AFI          0       agfi-01c6d8278eaa0f63f  loaded            0        ok               0       0x04261818
AFIDEVICE    0       0x1d0f      0xf001      0000:00:1d.0
sdacl-slave-timeout=0
virtual-jtag-slave-timeout=0
ocl-slave-timeout=0
bar1-slave-timeout=0
dma-pcis-timeout=0
pcim-range-error=0
pcim-axi-protocol-error=0
dma-range-error=0
pcim-axi-protocol-4K-cross-error=0
pcim-axi-protocol-bus-master-enable-error=0
pcim-axi-protocol-request-size-error=0
pcim-axi-protocol-write-incomplete-error=0
pcim-axi-protocol-first-byte-enable-error=0
pcim-axi-protocol-last-byte-enable-error=0
pcim-axi-protocol-bready-error=0
pcim-axi-protocol-rready-error=0
pcim-axi-protocol-wchannel-error=0
sdacl-slave-timeout-addr=0x0
sdacl-slave-timeout-count=0
virtual-jtag-slave-timeout-addr=0x0
virtual-jtag-slave-timeout-count=0
ocl-slave-timeout-addr=0x0
ocl-slave-timeout-count=0
bar1-slave-timeout-addr=0x0
bar1-slave-timeout-count=0
dma-pcis-timeout-addr=0x0
dma-pcis-timeout-count=0
pcim-range-error-addr=0x0
pcim-range-error-count=0
pcim-axi-protocol-error-addr=0x0
pcim-axi-protocol-error-count=0
pcim-write-count=456
pcim-read-count=776
DDR0
   write-count=0
   read-count=0
DDR1
   write-count=268435456
   read-count=0
DDR2
   write-count=268435456
   read-count=0
DDR3
   write-count=0
   read-count=0
Clock Group A Frequency (Mhz)
250  125  375  500
Clock Group B Frequency (Mhz)
250  125
Clock Group C Frequency (Mhz)
136  4
Power consumption (Vccint):
   Last measured: 12 watts
   Average: 12 watts
   Max measured: 12 watts
Cached agfis:
   agfi-01c6d8278eaa0f63f
   agfi-02309af29db2b3bf1
```

AWScsaralay, 2023-10-25:
Hi, Please note that the Shell-CL interface should be sync'ed to `clk_main_a0` as stated in [AWS_Shell_Interface_Specification.md#clocks](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#clocks). 

Could you please confirm if your design is driving the PCIM interface based off `clk_main_a0`? If not, I recommend adding a clock domain crossing logic between your AXI4 driver running on 136MHz (Clock Group C, based on your metrics above) and the `clk_main_a0` before feeding the PCIM interface. 

Please let use know if you need any details or have questions. 

Thanks!
Chakra

tewujianke, 2023-10-26:
Thank you for the response! The design does have clock domain crossing from the **clk_extra_c0** (CL clk) to **clk_main_a0**. **xpm_fifo_axif** is used to do this. We connect axi_in to our clk logic and axi_out to the shell, as shown in the below code snippet.

Another experiment I did was to remove the multi-clk domain design and drive both shell and CL with **clk_main_a0**; the same hanging behavior is observed at 136MHz, but it still worked fine with higher frequencies. We also noticed the design is doing b2b 64B read to the host (with max outstanding 32 req), and when it hangs, some read requests don't get responses. By inserting bubbles/gaps into the b2b read requests, the hanging would be gone. Any other advice would be much appreciated!

`module axi_asyncff (
    input                       wclk,
    input                       rclk,
    axi_bus_t.master axi_in,
    axi_bus_t.slave  axi_out,
    input                       rst_n
) ;

// xpm_fifo_axif: AXI Memory Mapped (AXI Full) FIFO
// Xilinx Parameterized Macro, version 2020.1

xpm_fifo_axif #(
   .AXI_ADDR_WIDTH(64),            // DECIMAL
   .AXI_ARUSER_WIDTH(1),           // DECIMAL
   .AXI_AWUSER_WIDTH(1),           // DECIMAL
   .AXI_BUSER_WIDTH(1),            // DECIMAL
   .AXI_DATA_WIDTH(512),            // DECIMAL
   .AXI_ID_WIDTH(16),               // DECIMAL
   .AXI_LEN_WIDTH(8),              // DECIMAL
   .AXI_RUSER_WIDTH(1),            // DECIMAL
   .AXI_WUSER_WIDTH(1),            // DECIMAL
   .EN_RESET_SYNCHRONIZER(1),
   .CDC_SYNC_STAGES(2),            // DECIMAL
   .CLOCKING_MODE("independent_clock"), // String
   .ECC_MODE_RDCH("no_ecc"),       // String
   .ECC_MODE_WDCH("no_ecc"),       // String
   .FIFO_DEPTH_RACH(16),         // DECIMAL
   .FIFO_DEPTH_RDCH(16),         // DECIMAL
   .FIFO_DEPTH_WACH(16),         // DECIMAL
   .FIFO_DEPTH_WDCH(16),         // DECIMAL
   .FIFO_DEPTH_WRCH(16),         // DECIMAL
   .FIFO_MEMORY_TYPE_RACH("auto"), // String
   .FIFO_MEMORY_TYPE_RDCH("auto"), // String
   .FIFO_MEMORY_TYPE_WACH("auto"), // String
   .FIFO_MEMORY_TYPE_WDCH("auto"), // String
   .FIFO_MEMORY_TYPE_WRCH("auto"), // String
   .PACKET_FIFO("false"),          // String
   .PROG_EMPTY_THRESH_RDCH(5),    // DECIMAL
   .PROG_EMPTY_THRESH_WDCH(5),    // DECIMAL
   .PROG_FULL_THRESH_RDCH(5),     // DECIMAL
   .PROG_FULL_THRESH_WDCH(5),     // DECIMAL
   .RD_DATA_COUNT_WIDTH_RDCH(5),   // DECIMAL
   .RD_DATA_COUNT_WIDTH_WDCH(5),   // DECIMAL
   .SIM_ASSERT_CHK(0),             // DECIMAL; 0=disable simulation messages, 1=enable simulation messages
   .USE_ADV_FEATURES_RDCH("0000"), // String
   .USE_ADV_FEATURES_WDCH("0000"), // String
   .WR_DATA_COUNT_WIDTH_RDCH(5),   // DECIMAL
   .WR_DATA_COUNT_WIDTH_WDCH(5)    // DECIMAL
)
xpm_fifo_axif_inst (
   .dbiterr_rdch(),             // 1-bit output: Double Bit Error- Indicates that the ECC
                                            // decoder detected a double-bit error and data in the FIFO core
                                            // is corrupted.

   .dbiterr_wdch(),             // 1-bit output: Double Bit Error- Indicates that the ECC
                                            // decoder detected a double-bit error and data in the FIFO core
                                            // is corrupted.

   .m_axi_araddr(axi_out.araddr),             // AXI_ADDR_WIDTH-bit output: ARADDR: The read address bus gives
                                            // the initial address of a read burst transaction. Only the
                                            // start address of the burst is provided and the control
                                            // signals that are issued alongside the address detail how the
                                            // address is calculated for the remaining transfers in the
                                            // burst.

   .m_axi_arburst(),           // 2-bit output: ARBURST: The burst type, coupled with the size
[... 376 lines cut ...]
                                            // of bytes.

   .s_axi_wlast(axi_in.wlast),               // 1-bit input: WLAST: Indicates the boundary of a packet.
   .s_axi_wstrb(axi_in.wstrb),               // AXI_DATA_WIDTH/8-bit input: WSTRB: The byte qualifier that
                                            // indicates whether the content of the associated byte of TDATA
                                            // is processed as a data byte or a position byte. For a 64-bit
                                            // DATA, bit 0 corresponds to the least significant byte on
                                            // DATA, and bit 0 corresponds to the least significant byte on
                                            // DATA, and bit 7 corresponds to the most significant byte. For
                                            // example: STROBE[0] = 1b, DATA[7:0] is valid STROBE[7] = 0b,
                                            // DATA[63:56] is not valid

   .s_axi_wuser(1'b0),               // AXI_WUSER_WIDTH-bit input: WUSER: The user-defined sideband
                                            // information that can be transmitted alongside the data
                                            // stream.

   .s_axi_wvalid(axi_in.wvalid)              // 1-bit input: WVALID: Indicates that the master is driving a
                                            // valid transfer. A transfer takes place when both WVALID and
                                            // WREADY are asserted

);

// End of xpm_fifo_axif_inst instantiation
endmodule
`

closed 2024-01-03 by tewujianke