aws/aws-fpga #625: Time delay in filling up BRAM using DMA

Gogul-N, 2023-10-12:
Hi,

I have created a 2 MB BRAM memory using Block Memory Generator. But in my test case, I want to store the data which is more than 2 MB (I may transfer up to 100 MB). So there is module which takes care of this transaction, which will feed the data from the BRAM to the receiver which I want to validate at 40 MHz speed. When half of the memory is feed to the receiver it will update the half count . When the half count is updated immediately half of the memory will be written with next 1 MB of new data. 

There is an AGC RAM in the receiver. When I configure the AGC RAM. This Half count reading is taking more time. Even filling up the memory is taking more time. So the receiver is not able to receive the actual data. When the AGC is not configured, there is no issue. Half count read is fast and filling up memory is also fast. What will be the cause of this issue?

The c code logic  I am using is

`int target_half = 0;
while (1) {
   rc = fpga_dma_burst_read(read_fd, (uint8_t*)(&half_count),4, (BASE_ADDR_IPC + 0x1c));

    // Check if half_count has changed
    if (half_count != target_half) {
      
  // Update the target_half variable
                target_half = half_count;
		printf("Half_count: %d\n",half_count);
        // Determine which half to fill based on the parity of half_count
        if (half_count % 2 == 1) {
            		rc = fpga_dma_burst_write(write_fd, &((uint8_t *)ptr_to_adc_buffer)[BUFFER_ADDR], half_size, BASE_ADDR_ADC);
			BUFFER_ADDR = BUFFER_ADDR + 1048576;
        } else {
            
			rc = fpga_dma_burst_write(write_fd, &((uint8_t *)ptr_to_adc_buffer)[BUFFER_ADDR], half_size, BASE_ADDR_ADC + 0x100000);
			BUFFER_ADDR = BUFFER_ADDR + 1048576;
        }
    }

       if (exp_half_count == half_count) {
	break;
    }`
    
    Note: clock_main_a0 is 125 MHz and the receiver receiving the data at 40 MHz. When I go with pci it is taking 1.3 ms to write the 40000 data into BRAM( I don't want this since read is happening at 1 ms inside the FPGA). So, I am going with DMA where it takes only 0.19 ms to write the data into BRAM.
        
    Kindly help me solving this issue. Thanks.

Gogul-N, 2023-10-16:
Hi,

Filling up the memory is done using DMA as well as the interrupt is also handled by DMA. Both operations concurrently using DMA causing the delay I think so. 

Thank you.

czfpga, 2023-10-23:
Hi @Gogul-N,

I need your help to better understand the issue you have, since it seems very closely related to your receiver design. 

It sounds like you have implemented a ping-pong style BRAM access mechanism that once the half of the BRAM data has been fed into the downstream receiver, the `half_count` will get updated and the other half of the memory will be used to provide data to the downstream receiver while the DMA starts filing up the first half memory again. 

And you have two such implementations: one for receiver input data and one for Rx's AGC memory? The issue you're seeing is that the AGC memory gets fed slower? Does that sound like a correct understanding? If not, can you elaborate more to clarify the issue you have?

Thanks,

Chen

Gogul-N, 2023-10-26:
Hi @czfpga 

Filling up AGC memory is not the problem. When it is not filled, the receiver is not functioning so my half_count is incremented without any issue. 

The actual problem is with the interrupt. Interrupt signals from my design was connected to irq_req[15:0]. Here the irq_req[15:0] is handled by DMA Polling. I was trying to listen to the interrupt as well as to fill up the memory inside FPGA using DMA simultaneously using threading function in C. Since both were trying to access the DMA simultaneously the delay was there, I think so.

Thank you.

closed 2023-11-22 by Gogul-N