aws/aws-fpga #629: Inserting ILA is improving the performance

Gogul-N, 2023-11-14:
Hi,

[image]

Look at the above image. From **module A**, a gated clock is going to **module B**. We found that **module B** is not functioning properly. So, inserted **ILA 1** in **module B** and found that gated clock is always high inside **module B**. To identify the root cause, we inserted **ILA 2** in **module A**. After inserting the **ILA 2** in **module A**, both modules are functioning as expected. And when we removed the **ILA 2** in **module A**, again the same problem faced in **module B**. 

What will be the cause of this issue? 
NOTE: We are using BUFGMUX (to replace OR gate) for gated clock. 

Thank you.

Gogul-N, 2023-11-15:
Hi @AWSjoeluc ,

Thank you for your reply.

While simulating the example design, I am getting the following errors.

```
ERROR: [USF-XSim-62] 'elaborate' step failed with error(s) while executing '/home/centos/src/project_data/aws-fpga/hdk/cl/examples/hello_world_hlx/example_projects/hello_world.sim/sim_1/behav/xsim/elaborate.sh' script. Please check that the file has the correct 'read/write/execute' permissions and the Tcl console output for any other possible errors or warnings.

ERROR: [Vivado 12-4473] Detected error while running simulation. Please correct the issue and retry this operation.

ERROR: [Common 17-39] 'launch_simulation' failed due to earlier errors.
```

Thank you

AWSjoeluc, 2023-11-15:
Hi @Gogul-N,

1. Does that file exist at the path in the message?
2. Did you clone the repo and run all prior steps with the same user account?
3. Can you try changing all the file permissions with `chmod -R ___ <top of hierarchy`?

If nothing works, can you try a fresh clone and follow [RTL_SIMULATING_CL_DESIGNS](https://github.com/aws/aws-fpga/blob/master/hdk/docs/RTL_Simulating_CL_Designs.md) to run the simulation without any changes?

Thanks,
-Joe

Gogul-N, 2023-11-20:
Hi @AWSjoeluc,

As per the solution given in #590 , I am able to simulate now.

Thank you.