aws/aws-fpga #630: Update

Hotwright, 2023-11-18:
There has been no update for 2 years. The latest  AMD/Xilinx is 2023.2 I think you support 2021.2.

Are there any plans to update this?

AWSjoeluc, 2023-11-20:
Hello!

Yes, we're actively working on a customer release to enable Vivado 2023.2.

Does Vivado 2023.2 (or any version since 2021.2) add any new features that you're hoping to use on F1?

Thanks,
-Joe

Hotwright, 2023-11-21:
Yes. 2023.2 Vitis has a whole new GUI and I find that each release fixes bugs and makes useability fixes. As a user I keep my AMD/Xilinx tools up to date and those releases are so big I can only have the latest release on my server. 

 

I’m willing to be a tester if that helps me get access to the latest release. 

 

I’m trying to port my latest work at Hotwright.com which is in 2023.1

 

Cheers

 

Steve

PS Something about me.

https://bit.ly/WhoIsSteveCasselman

ioeddk, 2023-12-01:
I'm also expecting support for the Vitis 2023 release. Specifically, there are bugs fixed for the Task library if we want to implement Task-Channel kernels. I believe the potential benefits are there to updating this repository to keep up with the newer version of Xilinx's development tools.

Hotwright, 2023-12-04:
Are task channels a way to have to concurrent process talk to each other? Altera had this kind of extension in their OpenCL compiler. 

 

FYI I’m giving a talk on Tuesday the 5th at 6 pm PST. The state machine is runtime loadable so you can change the behavior of your hardware at runtime. I have a demo with 2 threads running on the SoC and a tokenizer that is loadable at runtime. The two threads feed data and retrieve tokens respectively.  

 

https://www.svec.org/event/the-hotstate-machinea-runtime-loadable-microcoded-algorithmic-statemachine/

 

Cheers

 

Steve

augierg, 2024-03-20:
@AWSjoeluc : I am also looking forward to support for 2023.2. We are having issues getting our RTL to complete with a simple `synth_design -lint`  in 2021.2 because of crash with `Abnormal program termination (11)`. We face no such issue with 2023.2, and now we are at the stage we want to integrate our CL with the F1 FPGA shell. Having a support for 2023.2 would greatly help us save time in debugging the reason of a crash on a 2 year old version

jjones-github, 2024-03-20:
It would be great to get 2023.2 supported!

ioeddk, 2024-03-20:
Really need a newer version of Vitis AMI since there are a lot of bugs in the 2021.2 version. For example, using pragma dependence off to specify the false dependency, the synthesis tool would run into infinite loop printing "identified false dependence". This is a crucial feature of our project. Also, some times the use of a vector library would get "conflict pragma of array partition and aggregate", which is a behavior not documented anywhere.

AWSjoeluc, 2024-03-20:
Hi everyone, thanks for the data points and known issues with 2021.2. Learning about these issues really helps improve the devkit going forward.

We've been prioritizing the 2023.2 AMI release and will update you all as soon as it's ready!

augierg, 2024-07-02:
> Hi everyone, thanks for the data points and known issues with 2021.2. Learning about these issues really helps improve the devkit going forward.
> 
> We've been prioritizing the 2023.2 AMI release and will update you all as soon as it's ready!

@AWSjoeluc : is there an expected availability estimate for the 2023.2 AMI release ? We are still facing issues with Vivado 2021.2 not fully supporting some of systemverilog constructs, and crashing randomly when running synthesis on our RTL, and all these issues appear to be resolved in 2023.2

ioeddk, 2024-09-02:
As @augierg mentioned, is there an estimated expected availability for the new AMI release? The current AMI is CentOS7 based and it just went EOL at the end of June. Currently, this AMI can't be used because the CentOS7 doesn't allow me to install new software with `yum` by complaining `Cannot find a valid baseurl for repo: extras/7/x86_64`. This is a known issue a lot of people meet after the EOL.

augierg, 2024-09-10:
> ```
> 
> We appreciate everyone's patience with the long awaited release!

@AWSjoeluc : thanks for the update. Would you be able to give some timeframe for this long awaited release (3mo, 6mo, 1y, more ?)

AWSjoeluc, 2025-01-07:
Hello!

Sorry for the delayed response, you can start using the new Vivado 2024.1 [XDMA shell release](https://github.com/aws/aws-fpga/pull/670) and [small shell release](https://github.com/aws/aws-fpga/pull/669) now that their pull requests have been opened.

Additionally, the [f2 branch](https://github.com/aws/aws-fpga/tree/f2) has been added if you'd like to take advantage of the new F2 features on small shell.

closed 2025-04-21 by TrahStevAmz