aws/aws-fpga #631: Name port does not exist for instance f1_inst

kevinkiener, 2023-12-09:
Hello,
When trying to run the synthesis of the f1 core in HLX mode I am running into the errors:
```
[Synth 8-448] named port connection 'irq_req' does not exist for instance 'f1_inst' of module 'cl_f1_inst_0' ["/home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/synth/cl.v":1144]
[Synth 8-448] named port connection 'status_vdip' does not exist for instance 'f1_inst' of module 'cl_f1_inst_0' ["/home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/synth/cl.v":1259]
[Synth 8-448] named port connection 'status_vled' does not exist for instance 'f1_inst' of module 'cl_f1_inst_0' ["/home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/synth/cl.v":1260]
```
The error also occurs when trying to run the synthesis of the cl_hello_world_ref_hlx example.
What am I missing?
 
Thanks in advance!

AWScsaralay, 2023-12-11:
Hi, 

could you confirm if you setup HLX environment using instructions in [setup-hlx-environment](https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_hello_world_ref_hlx#setup-hlx-environment) and build DCP using instructions in [create-example-design-command-line](https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_hello_world_ref_hlx#create-example-design-command-line)?

Could you also share the vivado version in use? 

Thanks!
Chakra

kevinkiener, 2023-12-12:
Hi Chakra,

I can confirm that I am using the instructions setup-hlx-environment but I am using the GUI instructions with `aws::make_ipi`. 
Now I tried reinstalling the image and using a new folder for the aws setup. But now I get the following critical warning which breaks the synthesis:
`[Designutils 20-1280] Could not find module 'bd_c5e6_microblaze_I_0'. The XDC file /home/centos/src/project_data/aws-fpga-2/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_0/bd_c5e6_microblaze_I_0.xdc will not be read for any cell of this module.
`
Best,
Kevin

kevinkiener, 2023-12-15:
Hi,

I have tried to use a completely new installation of the [FPGA image](https://aws.amazon.com/marketplace/pp/prodview-gimv3gqbpe57k).
Then I did set up aws configure and installed a GUI as well as xrdp.
After that, I use the GUI via remote desktop connection and then I followed the cl_hello_world_ref_hlx guide
https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_hello_world_ref_hlx
However, with this approach I still get a failed synthesis and the errors as mentioned previously:
```
[Designutils 20-1280] Could not find module 'bd_c5e6_microblaze_I_0'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_0/bd_c5e6_microblaze_I_0.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'bd_c5e6_rst_0_0'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_1/bd_c5e6_rst_0_0_board.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'bd_c5e6_rst_0_0'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_1/bd_c5e6_rst_0_0.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'bd_c5e6_ilmb_0'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_2/bd_c5e6_ilmb_0.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'bd_c5e6_dlmb_0'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_3/bd_c5e6_dlmb_0.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'bd_c5e6_iomodule_0_0'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/bd_0/ip/ip_10/bd_c5e6_iomodule_0_0_board.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'ddr4_core_microblaze_mcs'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/ip/ddr4_core/ip_0/ddr4_core_microblaze_mcs_board.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'ddr4_core'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/ip/ddr4_core/par/ddr4_core.xdc will not be read for any cell of this module.
[Designutils 20-1280] Could not find module 'ddr4_core'. The XDC file /home/centos/src/project_data/aws-fpga/hdk/cl/examples/cl_hello_world_ref_hlx/example_projects/cl_hello_world_ref.gen/sources_1/bd/cl/ip/cl_f1_inst_0/ip/ddr4_core/ddr4_core_board.xdc will not be read for any cell of this module.

```

czfpga, 2024-03-05:
Hi Kevin,

We're currently having AMD involved in looking into this issue. We'll keep you updated on this issue as soon as we hear back from them. Thanks.

czfpga, 2024-03-05:
Just to confirm to see if you've tired commenting out this line at https://github.com/aws/aws-fpga/blob/863d963308231d0789a48f8840ceb1141368b34a/hdk/common/shell_v04261818/hlx/build/scripts/subscripts/make_post_synth_dcp.tcl#L90C21-L90C27 and rerunning the flow to see if the problem gets fixed. If not, please post the latest warning/errors shown on your end. Thanks.

kevinkiener, 2024-03-06:
With this fix the issue of not being able to run has been resolved it does work now. 
The critical warning remains but it works.
To fix the issue with the port not being available I had to set up a new clean instance.