aws/aws-fpga #635: Linux Kernel 6 support for xdma

Fischiii, 2024-02-05:
When building the xdma drivers with ubuntu 22.04.3 (Kernel 6.2) or greater a build failure is reported:

```
ou are using: gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
/tmp/aws-fpga/sdk/linux_kernel_drivers/xdma/Makefile:25: XVC_FLAGS: .
/tmp/aws-fpga/sdk/linux_kernel_drivers/xdma/libxdma.c: In function ‘transfer_destroy’:
/tmp/aws-fpga/sdk/linux_kernel_drivers/xdma/libxdma.c:2855:25: error: implicit declaration of function ‘pci_unmap_sg’; did you mean ‘pci_unmap_rom’? [-Werror=implicit-function-declaration]
2855 | pci_unmap_sg(xdev->pdev, sgt->sgl, sgt->nents,
| ^~~~~~~~~~~~
| pci_unmap_rom
```

Building with Ubuntu 22.04.2 or lower (Kernel 5.19) works as expected.

AWSjoeluc, 2024-02-06:
Hello!

Thank you for reaching out with this issue. Can you confirm if you're building the XDMA drivers with this commit from AMD/Xilinx adding support for kernel 6.X?

https://github.com/Xilinx/dma_ip_drivers/commit/9cdc9e335d51dc4e0e9d9fd6b4592ae55e30de72

Fischiii, 2024-02-07:
@AWSjoeluc thanks for the swift response, we just tested it and with the newest version of the driver it is building successfully.

AWSjoeluc, 2024-02-07:
Glad to hear it's successfully working! Do you have any more questions before resolving this this issue?

Fischiii, 2024-02-08:
No further questions. Thanks again.

closed 2024-02-08 by Fischiii