aws/aws-fpga #637: Files missing from AMI

jerhill, 2024-03-12:
I have a fresh install of the FPGA AMI ( FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc) and it appears to be missing some files.

I am unable to do the quickstart step 4 from src/README.md because there is no test directory in home/centos/src. The only things in src are README.md and the empty project_data directory. 

I ignored the missing tests directory problem, but then came to a crashing halt later when I received the error "You are not licensed to encrypt IEEE1735 Version 2" which, after some research, indicates that the file /home/centos/src/license/XilinxAWS.lic is also missing. To be sure I didn't break anything myself, I started a fresh instance from the AMI, and the files are not present.

Have these files been intentionally removed? If so, how do I access the license file (and whatever else might have moved or been deleted?)

Thanks for your assistance.

-JMH

AWSjoeluc, 2024-03-14:
Hello!

Thank you so much for reaching out with this issue. We can confirm that step 4 of those instructions will fail due to the missing files. However, I'm not able to reproduce the failure you're seeing with regard to the license file.

I suspect that you may be running an instance with IMDSv2. Currently, Vivado 2021.2 only supports IMDSv1. We plan to add support for Vivado 2023.2 with IMDSv2.

Can you please share the full steps you took to reproduce the license issue?

Thanks again!

jerhill, 2024-03-14:
I am indeed running with IMDSv2, as per our institutional instructions. I never considered that this would have any effect on the software I was working with. I'll have to see if I can use v1.

I hit this problem while working through the [FPGA development workshop](https://fpga-development-on-ec2.workshop.aws/en/) . The error popped up [here](https://fpga-development-on-ec2.workshop.aws/en/4-f1-application-development-flow/hdk-development-flow.html#hdk-development-flow-exercise) in the Hello World example when trying to run the _aws_build_dcp_from_cl.sh_ script.

I eagerly anticipate the release of a 2023.2 version. :)

Thanks for a prompt response!

-JMH