aws/aws-fpga #640: Accessing the AXI-Lite via the fpga_pci_attach call...

wpiman, 2024-04-26:
I am able to use the fpga_pci.h and fpga_mgmt.h libraries in my code just fine when I run the code as "sudo".  
When I run it as a user, I get an error... 

It is a simple AXI-LITE peek/poke command.  Here is the C code that offends...  

```
 rc = fpga_pci_attach(slot_id, FPGA_APP_PF, APP_PF_BAR0, 0, &pci_bar_handle);
 fail_on(rc, out, "Unable to attach to the AFI on slot id %d", slot_id);
```

And the error when run as a non-root user....

> "2024-04-26T21:33:40.200320Z, undefined, ERROR, peekPoke.c +128: main(): Unable to attach to the AFI on slot id 0"

I followed these instructions to run as a non-root user... 

[(https://github.com/aws/aws-fpga/blob/master/sdk/README.md#using-fpga-as-non-root-user)]

[ec2-user@ip-10-20-7-111 peekPoke]$ env|grep AWS_FPGA
AWS_FPGA_SDK_GROUP=fpga-user
AWS_FPGA_ALLOW_NON_ROOT=y
AWS_FPGA_REPO_DIR=/home/ec2-user/aws-fpga
BASH_FUNC_allow_non_root()=() {  [ ! -z ${AWS_FPGA_ALLOW_NON_ROOT} ]
BASH_FUNC_allow_others()=() {  [ ! -z ${AWS_FPGA_SDK_OTHERS} ]

These allow me to run the "fpga-load-local-image" as a user, as well as the "fpga-set-virtual-dip-switch" command just fine.  However, the code that I wrote still throw that error.

czfpga, 2024-05-06:
Hi,

Thank you for reaching out. I want to let you know that we're currently investigating this issue. We will keep you updated.

s03311251, 2024-06-28:
I am also having an almost identical problem, where I can't run peek/poke command without `sudo`. Here is what I have commanded:

```bash
newgrp fpgauser
sudo usermod -a -G fpgauser $(whoami)
export AWS_FPGA_ALLOW_NON_ROOT=y
export AWS_FPGA_SDK_OTHERS=y
cd $AWS_FPGA_REPO_DIR && source sdk_setup.sh
cd ~/my_project_folder && make all
./my_project_executable
```

and below is the error message when I set CONFIG_LOGLEVEL to 4:

<details>

```log
2024-06-28T13:44:50.935611Z, undefined, INFO, fpga_pci.c +281: fpga_pci_init(): FPGA_PCI_BARS_MAX=64
2024-06-28T13:44:50.935825Z, undefined, DEBUG, fpga_pci_sysfs.c +62: fpga_pci_get_id(): Error opening /sys/bus/pci/devices/./vendor: error_number=2
2024-06-28T13:44:50.935832Z, undefined, DEBUG, fpga_pci_sysfs.c +285: fpga_pci_get_resource_map_ids(): Error retrieving vendor_id
2024-06-28T13:44:50.935841Z, undefined, DEBUG, fpga_pci_sysfs.c +62: fpga_pci_get_id(): Error opening /sys/bus/pci/devices/../vendor: error_number=2
2024-06-28T13:44:50.935846Z, undefined, DEBUG, fpga_pci_sysfs.c +285: fpga_pci_get_resource_map_ids(): Error retrieving vendor_id
2024-06-28T13:44:50.936590Z, undefined, DEBUG, fpga_pci_sysfs.c +188: fpga_pci_get_pci_resource_info(): stat failed, path=/sys/bus/pci/devices/0000:00:1e.0/resource1: error_number=2
2024-06-28T13:44:50.936598Z, undefined, DEBUG, fpga_pci_sysfs.c +239: fpga_pci_get_resources(): Unable to read resource information for 1
2024-06-28T13:44:50.936821Z, undefined, DEBUG, fpga_pci.c +171: fpga_pci_bar_attach(): enter
2024-06-28T13:44:50.936827Z, undefined, INFO, fpga_pci.c +186: fpga_pci_bar_attach(): vendor_id=0x1d0f, device_id=0x1041, DBDF:0000:00:1e.0, resource_num=0, size=16384, burstable=1
2024-06-28T13:44:50.936833Z, undefined, DEBUG, fpga_pci.c +141: fpga_pci_check_file_id(): path=/sys/bus/pci/devices/0000:00:1e.0/vendor
2024-06-28T13:44:50.936857Z, undefined, DEBUG, fpga_pci.c +141: fpga_pci_check_file_id(): path=/sys/bus/pci/devices/0000:00:1e.0/device
2024-06-28T13:44:50.936903Z, undefined, DEBUG, fpga_pci.c +240: fpga_pci_bar_attach(): Opening sysfs_name=/sys/bus/pci/devices/0000:00:1e.0/resource0
2024-06-28T13:44:50.936932Z, undefined, DEBUG, fpga_pci.c +95: fpga_pci_bar_alloc(): enter
2024-06-28T13:44:50.936937Z, undefined, DEBUG, fpga_pci.c +104: fpga_pci_bar_alloc(): allocated handle=0
2024-06-28T13:44:50.936941Z, undefined, DEBUG, fpga_pci.c +59: fpga_pci_bar_set_mem_base_size(): handle=0
2024-06-28T13:44:50.936945Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.936952Z, undefined, DEBUG, fpga_hal_mbox.c +55: fpga_hal_mbox_init(): enter
2024-06-28T13:44:50.936955Z, undefined, DEBUG, fpga_hal_mbox.c +43: fpga_hal_mbox_print_reg_offsets(): FMB_BASE=0x00000000, status=0x0000000c
2024-06-28T13:44:50.936959Z, undefined, DEBUG, fpga_hal_mbox.c +46: fpga_hal_mbox_print_reg_offsets(): Reg Offsets: mb_wr_index=0x00000020, mb_wr_data=0x00000024, mb_wr_len=0x00000028
2024-06-28T13:44:50.936964Z, undefined, DEBUG, fpga_hal_mbox.c +49: fpga_hal_mbox_print_reg_offsets(): Reg Offsets: mb_rd_index=0x00000030, mb_rd_data=0x00000034, mb_rd_len=0x00000038
2024-06-28T13:44:50.936969Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
2024-06-28T13:44:50.936973Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.936977Z, undefined, DEBUG, fpga_pci.c +395: fpga_pci_peek(): handle=0, offset=0xc, value=0x00000000
2024-06-28T13:44:50.936982Z, undefined, DEBUG, fpga_hal_mbox.c +301: fpga_hal_mbox_write(): enter
2024-06-28T13:44:50.936986Z, undefined, DEBUG, fpga_hal_mbox.c +230: fpga_hal_mbox_write_async(): enter
2024-06-28T13:44:50.936990Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
2024-06-28T13:44:50.936994Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.936998Z, undefined, DEBUG, fpga_pci.c +395: fpga_pci_peek(): handle=0, offset=0xc, value=0x00000000
2024-06-28T13:44:50.937002Z, undefined, DEBUG, fpga_pci.c +339: fpga_pci_poke(): handle=0, offset=0x20, value=0x00000000
2024-06-28T13:44:50.937006Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
2024-06-28T13:44:50.937009Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.937012Z, undefined, DEBUG, fpga_pci.c +339: fpga_pci_poke(): handle=0, offset=0x24, value=0x00010002
2024-06-28T13:44:50.937016Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
2024-06-28T13:44:50.937019Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.937022Z, undefined, DEBUG, fpga_pci.c +339: fpga_pci_poke(): handle=0, offset=0x24, value=0x00000002
2024-06-28T13:44:50.937026Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
2024-06-28T13:44:50.937030Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.937033Z, undefined, DEBUG, fpga_pci.c +339: fpga_pci_poke(): handle=0, offset=0x24, value=0x00000004
2024-06-28T13:44:50.937037Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
2024-06-28T13:44:50.937042Z, undefined, DEBUG, fpga_pci.c +46: fpga_pci_bar_get(): handle=0
2024-06-28T13:44:50.937046Z, undefined, DEBUG, fpga_pci.c +339: fpga_pci_poke(): handle=0, offset=0x24, value=0x6b8b4567
2024-06-28T13:44:50.937050Z, undefined, DEBUG, fpga_pci.c +77: fpga_pci_bar_get_mem_at_offset(): handle=0
[... cut ...]
2024-06-28T13:44:50.960340Z, undefined, DEBUG, fpga_pci.c +240: fpga_pci_bar_attach(): Opening sysfs_name=/sys/bus/pci/devices/0000:00:1d.0/resource1
2024-06-28T13:44:50.960352Z, undefined, ERROR, fpga_pci.c +243: fpga_pci_bar_attach(): open failed: error_number=13
2024-06-28T13:44:50.960357Z, undefined, ERROR, src/main.c +181: peek_poke_example(): Unable to attach to the AFI on slot id 0
2024-06-28T13:44:50.960360Z, undefined, ERROR, src/main.c +332: main(): peek-poke example failed
```

</details>