aws/aws-fpga #644: Connect MicroBlaze Debug Module (MDM) to Virtual JTAG in Vivado IP Integrator

s03311251, 2024-06-12:
I have posted my problem in <https://github.com/aws/aws-fpga/issues/641#issuecomment-2115064425>, but I realise that it seems to be a different issue, as in the implemented design, the MDM is not connected to anything:

[image]

Therefore I would like to know the correct configuration to connect to MDM.

Below are what I originally posted in #641

-----

I hope this is the right place to post here, as I am expericing a similar problem:

Right now I'm working to put a MicroBlaze core with Vivado IP Integrator flow, however, I can't connect to the MicroBlaze Debug Module (MDM) when I deployed in on an EC2 F1 instance.

My design is as following:
[image]  
design files: [aws_mb_example.zip](https://github.com/aws/aws-fpga/files/15334751/aws_mb_example.zip)

I have followed <https://github.com/aws/aws-fpga/issues/507>, which mentioned to use "EXTERNAL HIDDEN" for BSCAN in MDM: 
[image]

However, it seems failed to identify MDM, as when I connect it with XSCT, I got the following: 
[image]  
Top left is the SSH session of EC2 F1, which was running `hw_server` and `sudo fpga-start-virtual-jtag -P 10201 -S 0`  
Bottom left is the Vivado tcl shell, connected to the Virtual JTAG on F1  
Bottom right is the XSCT, also connected to the Virtual JTAG on F1  
The commands are run in this order: SSH > Vivado > XSCT

The targets command in XSCT gives 8-digit numbers only, but supposedly it should gives something like "MicroBlaze Debug Module".

May I ask if there is any problem with my design with a MicroBlaze? Is there an alternative method to connect to MDM, or is there an example design for MicroBlaze that worked with AWS EC2?

Thank you.

czfpga, 2024-06-25:
Hi,

Thank you for reaching out. We don't currently have an example design with a MicroBlaze embedded. We're still investigating this issue internally and will keep you posted for any update we have.

s03311251, 2024-06-25:
If you want to replicate the problem, I have attached my design files above ([aws_mb_example.zip](https://github.com/aws/aws-fpga/files/15334751/aws_mb_example.zip))

s03311251, 2024-10-30:
Our project is still suffering from this problem, and for the sake of informing other people suffering the same problem, I have made another failed attempt, and the results was that, nothing in our design (including ILA & MDM) can be connected via Virtual JTAG.

Below is what I changed in this failed attempt:

- manually import `cl_debug_bridge` into the design sources, and re-customize the IP to set `BSCAN Master Count` to 1
- connected the BSCAN interface of MDM (`m0_bscan`) to `cl_debug_bridge` as following:  
  [image]
- copied `cl_top.sv`, `cl_ports_hlx.vh` and `sh_connectors_customs.vh` into the project and added the following to `cl_top.sv`:

  ```verilog
    cl cl_inst (
    // Users optionally add new BD external port connections here.
    // Terminate user-defined port connections list with a trailing comma.

    // Do not remove the following mandatory port connections.
    `include "sh_connectors_customs.vh"
    ,

    .m0_bscan_bscanid_en(m0_bscan_bscanid_en),
    .m0_bscan_capture(m0_bscan_capture),
    .m0_bscan_drck(m0_bscan_drck),
    .m0_bscan_reset(m0_bscan_reset),
    .m0_bscan_sel(m0_bscan_sel),
    .m0_bscan_shift(m0_bscan_shift),
    .m0_bscan_tck(m0_bscan_tck),
    .m0_bscan_tdi(m0_bscan_tdi),
    .m0_bscan_tdo(m0_bscan_tdo),
    .m0_bscan_tms(m0_bscan_tms),
    .m0_bscan_update(m0_bscan_update)
    );

  // Users optionally add custom RTL logic here.
  cl_debug_bridge CL_DEBUG_BRIDGE (
    .clk(clk_main_a0),
    .S_BSCAN_drck(drck),
    .S_BSCAN_shift(shift),
    .S_BSCAN_tdi(tdi),
    .S_BSCAN_update(update),
    .S_BSCAN_sel(sel),
    .S_BSCAN_tdo(tdo),
    .S_BSCAN_tms(tms),
    .S_BSCAN_tck(tck),
    .S_BSCAN_runtest(runtest),
    .S_BSCAN_reset(reset),
    .S_BSCAN_capture(capture),
    .S_BSCAN_bscanid_en(bscanid_en),
    .m0_bscan_bscanid_en(m0_bscan_bscanid_en),  // output wire m0_bscan_bscanid_en
    .m0_bscan_capture(m0_bscan_capture),        // output wire m0_bscan_capture
    .m0_bscan_drck(m0_bscan_drck),              // output wire m0_bscan_drck
    .m0_bscan_reset(m0_bscan_reset),            // output wire m0_bscan_reset
    .m0_bscan_runtest(),        // output wire m0_bscan_runtest
    .m0_bscan_sel(m0_bscan_sel),                // output wire m0_bscan_sel
    .m0_bscan_shift(m0_bscan_shift),            // output wire m0_bscan_shift
    .m0_bscan_tck(m0_bscan_tck),                // output wire m0_bscan_tck
    .m0_bscan_tdi(m0_bscan_tdi),                // output wire m0_bscan_tdi
    .m0_bscan_tdo(m0_bscan_tdo),                // input wire m0_bscan_tdo
    .m0_bscan_tms(m0_bscan_tms),                // output wire m0_bscan_tms
    .m0_bscan_update(m0_bscan_update)          // output wire m0_bscan_update
  );
  ```

AWSjoeluc, 2025-01-14:
Hi,

Thank you for your patience on this issue. We were able to successfully reproduce this issue on F1 with the previous Vivado v2021.2 release. We have not yet reproduced this issue with the new [Vivado v2024.1 release](https://github.com/aws/aws-fpga/tree/f1_xdma_shell). We're currently trying to replicate this on [F2](https://github.com/aws/aws-fpga/tree/f2) and hope to have an update soon!