aws/aws-fpga #645: phys_opt_design error during implementation

Gogul-N, 2024-07-06:
Hi,

I am getting the following error during phys_opt_design stage.
And I need to have phys_opt_design during implementation to overcome the hold violations since my design is big.

`ERROR: [Place 30-838] The following clock nets need to use the same clock routing resource, as their clock buffer sources are locked to sites that use the same routing track. One or more loads of these clocks are locked to clock region(s) X2Y12 X2Y13 X2Y14 X3Y12 X3Y13 X3Y14 X4Y13 X4Y14 X5Y14 which causes the clock partitions for these clocks to overlap. This creates unresolvable contention on the clock routing resources. If the clock buffers need to be locked, we recommend users constrain them to a clock region and not to specific BUFGCE/BUFG_GT sites so they can use different routing resources. If clock sources should be locked to specific BUFGCE/BUFG_GT sites that share the same routing resources, make sure loads of such clocks are not constrained to the same region(s). Clock nets sharing routing resources:
WRAPPER_INST/CL/cl_inst/xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx/u_clken_resync_fdo_clk/i_2t_reg_0
WRAPPER_INST/SH/kernel_clks_i/clkwiz_kernel_clk1/inst/CLK_CORE_DRP_I/clk_inst/clk_out2`

Kindly help me to solve this

Thank you.

ljp0101, 2024-07-07:
Pretty sure it's a Vivado bug. Good luck; I gave up.

https://github.com/aws/aws-fpga/issues/626

kboshear, 2024-07-16:
for place_design add the following switch -no_bufg_opt.  This allows for no BUFG to be added

closed 2024-07-30 by Gogul-N