aws/aws-fpga #647: fail in get_f1_ami_id()

huchensong, 2024-08-05:
i have a fresh install of the FPGA AMI ( FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc)and we are facing some problems in generateing bitstream.
when i run "firesim buildbitstream" command, it successfully Builds Verilog for FireSimWithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.Sha3RocketConfig-F65MHz_BaseF1Config and FPGA driver for FireSimWithNIC_DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimHighPerfConfigTweaks_chipyard.Sha3RocketConfig-F65MHz_BaseF1Config.
But it encoutners fatal error when trying to get ami id.(below is the error message in log files)
[image]
i try to solve the problem from the web,and the https://github.com/CSL-KU/firesim-nvdla/issues/1 is likely to be the same.
i follow the link, create a py file to test if the ami id is avaiable, but it fails.
[image]
By the way, i use a manager instance of z1d,2xlarge,which is also I launch bitstream on.
what can i do now ?

AWSjoeluc, 2024-08-05:
Hello!

Thank you so much for reaching out with this issue. Currently the AWS CLI and boto3 is not working when run from an instance using the `FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc`. When I run the same script from a separate machine I see no errors:

**COMMAND**

```python3
import boto3
import botocore

f1_ami_name = "FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc"

def get_f1_ami_id():
    client = boto3.client('ec2')
    response = client.describe_images(Filters=[{'Name': 'name', 'Values': [f1_ami_name]}])
    assert len(response['Images']) == 1
    print(response['Images'])

if __name__ == '__main__':
    get_f1_ami_id()
```

**STDOUT**

```
[{'Architecture': 'x86_64', 'CreationDate': '2023-02-01T20:33:08.000Z', 'ImageId': 'ami-0e178635787eb5e00', 'ImageLocation': 'aws-marketplace/FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc', 'ImageType': 'machine', 'Public': True, 'OwnerId': '679593333241', 'PlatformDetails': 'Linux/UNIX', 'UsageOperation': 'RunInstances', 'ProductCodes': [{'ProductCodeId': '3sppv90yg5o5uyuw20at6ywak', 'ProductCodeType': 'marketplace'}], 'State': 'available', 'BlockDeviceMappings': [{'DeviceName': '/dev/sda1', 'Ebs': {'DeleteOnTermination': True, 'Iops': 3000, 'SnapshotId': 'snap-05a37a5e6feb931e2', 'VolumeSize': 120, 'VolumeType': 'gp3', 'Throughput': 125, 'Encrypted': False}}, {'DeviceName': '/dev/sdb', 'Ebs': {'DeleteOnTermination': False, 'SnapshotId': 'snap-0b14714457ddb8533', 'VolumeSize': 5, 'VolumeType': 'gp2', 'Encrypted': False}}], 'Description': 'Xilinx 2021.2 release with latest AMI updates', 'EnaSupport': True, 'Hypervisor': 'xen', 'ImageOwnerAlias': 'aws-marketplace', 'Name': 'FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc', 'RootDeviceName': '/dev/sda1', 'RootDeviceType': 'ebs', 'VirtualizationType': 'hvm', 'DeprecationTime': '2025-02-01T20:33:08.000Z'}]
```

We recommend only launching new FireSim builds from a machine that is not running `FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc`. Do you see any potential issue with not launching new FireSim builds in this manner?

Thanks!
-Joe

huchensong, 2024-08-06:
Do you mean that I should run this command from a machine running a image ,which is different from my image:FPGA Developer AMI - 1.12.2-40257ab5-6688-4c95-97d1-e251a40fd1fc.
So the recommend step is ,recreating a instance with a new image?
if So, what is the recommend image?
This is what I see in firesim Docs ,latest version ,it recommends the image, i don't know which image to use.
[image]

AWSjoeluc, 2024-08-06:
Due to Centos end of life, some AWS CLI commands are not working on `FPGA Developer AMI - 1.12.2`. We're currently working on a new AMI release to resolve such issues. I would contact the FireSim maintainers on a way to skip the AMI verification check you're failing on above. Since you're confident you're using the correct AMI, the verification check should not be required.

closed 2025-04-21 by TrahStevAmz