aws/aws-fpga #648: ETA on new dev AMI?

ljp0101, 2024-08-07:
Or do you have instructions on how to get the Vivado GUI to work starting from scratch now that Centos 7 is EOL. I abandoned efforts after redirecting yum to the vault mirrors when GNOME didn't want to install and am hoping someone else solves this before I need to see waveforms.

ljp0101, 2024-08-07:
Does the Vivado licensing scheme require the dev AMI or can I use downloaded binaries from Xilinx on any AWS instance?

kristopk, 2024-12-06:
Hi @ljp0101 
The developer AMI based on Centos went EOL at the end of June 2024 and we can no longer update this AMI with the required patches. Feel free to use an alternative Amazon Linux 2 based Developer AMI found on AWS Marketplace: https://aws.amazon.com/marketplace/pp/B08NTMMZ7X
We plan to release a new Developer AMI and DevKit in near future with support for AMD tools 2024.1 (or newer) using Ubuntu OS.
The license is integrated into the developer AMI that is accessed via AWS Marketplace.
Thanks!

closed 2024-12-06 by kristopk