aws/aws-fpga #653: Support for newer version of Vivado...

wpiman, 2024-09-24:
The latest version in the repo for Vivado version support is 2021.2.  I tried changing this to 2024.1 and I was able to get through the hdk_setup but when building I get the following error....

ERROR: [Common 17-1404] Cannot decrypt IEEE-1735 envelope: key value doesn't match key name "xilinxt_2017_05" (perhaps the public key was incorrect at encryption time). [/aws-fpga/hdk/common/shell_v04261818/design/sh_ddr/synth/flop_ccf.sv:0]

Apparently this is a known issue and not resolvable.  

Is AWS still supporting this platform?  There are known issues with older version of Vivado we are trying to avoid....

https://adaptivesupport.amd.com/s/question/0D54U00007inRL7SAM/how-to-resolve-undef-cannot-decrypt-ieee1735-envelope-key-value-doesnt-match-key-name-xilinxt201705-perhaps-the-public-key-was-incorrect-at-encryption-time-in-vivado-2023?language=en_US

czfpga, 2025-01-02:
Hi @wpiman, we are updating to Vivado 2024.1 for F1. Please refer to [PR#669 ](https://github.com/aws/aws-fpga/pull/669)and [PR#670](https://github.com/aws/aws-fpga/pull/670). Feedback is appreciated.

czfpga, 2025-01-17:
@wpiman F1 2024.1 update has been released on[ f1_xdma_shell](https://github.com/aws/aws-fpga/tree/f1_xdma_shell) and [f1_small_shell](https://github.com/aws/aws-fpga/tree/f1_small_shell) branches. The F1 Ubuntu Developer AMI with 2024.1 Developer AMI has been deployed to all supported regions.

closed 2025-04-21 by TrahStevAmz