aws/aws-fpga #657: Compatibility of U200-Designed xclbin with AWS F1 Instance and XRT API Usage

JinB1, 2024-12-05:
Hello,

I synthesized a Verilog/SystemVerilog project designed for U200 using the xilinx_aws-vu9p-f1_shell-v04261818_201920_3.xpfm platform to generate an xclbin file. I then used /home/centos/aws-fpga/Vitis/tools/create_vitis_afi.sh to create an AFI from the xclbin file. Will this work without issues on an F1 instance with the vu9p platform?

So far, I’ve successfully created the AFI and uploaded it to the F1 instance. Currently, I’m testing to confirm if it operates as expected.

Additionally, I’d like to know if there are any potential issues when using XRT APIs, such as xrtKernelWriteRegister, on the FPGA.

Thank you.

kristopk, 2024-12-06:
Hi @JinB1 
Yes, FPGA developers can seamlessly migrate between on-prem (U200) and F1 platforms to enable scaling in AWS cloud. Glad to hear you are trying our the migration path.  Sounds like you have reviewed the documentation already:
HDK:
https://github.com/aws/aws-fpga-f1-u200/blob/main/README.md
https://github.com/aws/aws-fpga/blob/master/hdk/docs/U200_to_F1_migration_HDK.md
Vitis:
https://github.com/aws/aws-fpga/blob/master/Vitis/docs/Alveo_to_AWS_F1_Migration.md

We are not aware of any issues, with XRT APIs.  Please let us know if your experiencing issues so we can help.  
Thanks!

JinB1, 2024-12-07:
Thank you for your response

I am encountering an issue with xrtXclbinGetXSAName returning an error after loading the awsxclbin file.

Here is the relevant part of my code:

cpp
Copy code
printf("kernel opened: %s\n", KERNEL_NAME);
printf("device: %p\n", xrtDevice);

int device_name_size;
xrtXclbinGetXSAName(xrtDevice, nullptr, 0, &device_name_size);

std::vector<char> device_name(device_name_size);
xrtXclbinGetXSAName(xrtDevice, device_name.data(), device_name_size, nullptr);
printf("device name: %s\n", device_name.data());
Output:

yaml
Copy code
open device connection  
XRT_XCLBIN_PATH=/home/centos/vortex_afu.awsxclbin  
XRT_DEVICE_INDEX=0  
device opened: 0  
passed device open, xclbin load test  
xclbin loaded: /home/centos/vortex_afu.awsxclbin  
xclbin uuid: 7414593be8e5fb6037e11d1889a63412  
kernel opened: vortex_afu  
device: 0x55ed55cb94f0  
XRT build version: 2.12.0  
Build hash: 723d9e7abbe3a2c374682dbb1a59c47f230f3ee2  
Build date: 2022-01-11 19:12:26  
Git branch: 2021.2  

[XRT] ERROR: No such xclbin handle: Invalid argument  
[XRT] ERROR: No such xclbin handle: Invalid argument  
device name: (null)  
terminate called after throwing an instance of 'std::logic_error'  
what():  basic_string: construction from null is not valid  
Issue:
The kernel vortex_afu loads successfully, and the XCLBIN file seems valid (UUID is printed).
However, the call to xrtXclbinGetXSAName fails and returns "No such xclbin handle: Invalid argument".
Question:
What could cause xrtXclbinGetXSAName to fail even after a successful xrtDeviceLoadXclbinFile?
Is there any additional setup or step required to retrieve the XSA name in this case?