aws/aws-fpga #658: Inquiry About FPGA Developer AMI Support and Future Updates

JinB1, 2024-12-06:
Dear AWS FPGA Team,

I received a notice stating that the “FPGA Developer AMI” will no longer be offered to new subscribers as of December 5, 2024. As a current subscriber, I have the following questions regarding this change:
	1.	How long will the FPGA Developer AMI remain supported for existing subscribers?
	2.	For new users, does this mean F1 instances will need to be used without the FPGA Developer AMI? If so, will this create any challenges for development or deployment workflows?
	3.	Will the FPGA Developer AMI continue to receive updates, including maintenance, bug fixes, or feature enhancements, for existing subscribers?
	4.	Does this change imply any shift in AWS’s long-term strategy for FPGA-related services?

Thank you for your time and support. I look forward to your clarification on these points.

kristopk, 2024-12-06:
Hi @JinB1 
The developer AMI based on Centos went EOL at the end of June 2024 and we can no longer update this AMI with the required patches.  Feel free to use an alternative Amazon Linux 2 based Developer AMI found on AWS Marketplace:  https://aws.amazon.com/marketplace/pp/B08NTMMZ7X
We plan to release a new Developer AMI and DevKit in near future with support for AMD tools 2024.1 (or newer) using Ubuntu OS.  
Thanks!

JinB1, 2024-12-08:
Thanks for the update and the information!

Best regards,
JinB1

itaibak, 2024-12-09:
@kristopk What's the ETA for the new Ubuntu OS AMI release ?

ljp0101, 2024-12-10:
Problem with the AL2 AMI is that it's a Vivado version downgrade from the Centos AMI and there's no way to downgrade IPs so projects aren't directly portable.

nindanaoto, 2024-12-14:
I found that AWS F2 is supported by AMI based on Ubuntu, but I don't know whether this supports F1 or not. 
https://aws.amazon.com/marketplace/pp/prodview-f5kjsenkfkz5u

kristopk, 2024-12-14:
@nindanaoto @itaibak @ljp0101 F1 Devkit support for the [Ubuntu AMI](https://aws.amazon.com/marketplace/pp/prodview-f5kjsenkfkz5u) is coming very soon.  We will update this thread once we have a release candidate branch.

koceja, 2024-12-31:
Hi, I was wondering when the F2 instances will be supported by the FPGA Developer AMI (Ubuntu). Currently it says both F2 instance types do not support this AMI.

czfpga, 2024-12-31:
@koceja, F2 instances are fully supported by the FPGA Developer AMI (Ubuntu). You can find the detailed AMI information [here](https://github.com/aws/aws-fpga/blob/f2/User_Guide_AWS_EC2_FPGA_Development_Kit.md#fpga-developer-ami).  

Can you please point me to where it states that F2 instance types are not supported?

koceja, 2025-01-01:
Thanks Chen, please let me know when they are enabled.

czfpga, 2025-01-02:
@JinB1 @itaibak @ljp0101 @nindanaoto @koceja we are adding support for the Ubuntu Developer AMI to F1. Please refer to [PR#669 ](https://github.com/aws/aws-fpga/pull/669)and [PR#670](https://github.com/aws/aws-fpga/pull/670). Feedback is appreciated.

koceja, 2025-01-07:
@czfpga just following up here, is there an estimated time for enabling the AMI for F2 instances?

czfpga, 2025-01-07:
@koceja all I know it'll be coming very soon, but no more details other than that. But I will ensure to get you updated here as soon as the AMI is ready.

AWSjoeluc, 2025-01-09:
Hi @koceja, we've officially enabled the [FPGA Developer AMI (Ubuntu)](https://aws.amazon.com/marketplace/pp/prodview-f5kjsenkfkz5u) to be launched on F2 instances. Please reach out if you run into any issues!

koceja, 2025-01-10:
Thanks @AWSjoeluc and @czfpga, I really appreciate the help!

czfpga, 2025-01-17:
@koceja @nindanaoto @ljp0101 @itaibak @JinB1 F1 2024.1 Developer AMI has been enabled in all supported regions. Please let us know if you have any question.

itaibak, 2025-03-16:
@czfpga The new AMI is based on Ubuntu 20.04, which reaches EOL in May 2025. Are there any plans to upgrade to a newer Ubuntu version?