aws/aws-fpga #661: Request for elaboration in the F2 C++ flow readme file

yk-choi-yk, 2024-12-19:
Hi. 
First of all, thanks so much for the F2 update. I have been looking forward to this for some time.

Would it be possible if you could add instructions for 
1. building the host executable, 
2. building the xclbin file, 
3. generating the  AFI 

to the C++ flow described in https://github.com/aws/aws-fpga/blob/f2/vitis/README.md ?

Thanks,
- Young

kristopk, 2024-12-20:
Hi Young, 
Thanks for your feedback!  The current devkit (v2.0.0) supports Vitis hw emulation only.  Coming soon we will enable multiple shells including full Vitis support on F2 instances.
Thanks

yk-choi-yk, 2024-12-20:
Thanks so much for your reply!
Do you have approximate target date for C++ flow F2 execution?
I have been running a class on F1 for some time, and since F1 AMI is currently offline, I was wondering whether I should wait until F2 becomes available (while running hw_emu only), or look for non-AWS FPGA server for this winter quarter.

AWSNB, 2024-12-20:
Hi Young-kyu choi

What your ideal time you need this ?
Could you share which university/class is this and expected number of concurrent students ?

We’ll try to priority it on our side and update once we have the data
-Nafea

yk-choi-yk, 2024-12-20:
My mistake - it appears students can still access F1 through FPGA Developer AMI (Amazon Linux 2).
We'll go with this flow for this quarter, and I will try again the new F2 for spring (March, 2025).
Thanks so much for your offer, Nafea :)
-Young

closed 2024-12-20 by yk-choi-yk