aws/aws-fpga #671: F2 XDMA shell availability

itaibak, 2025-01-06:
I am looking forward to the migration of our product to the F2 platform, but we currently depend on the XDMA Shell. Could you please share the timeline for its release?

Additionally, I have a question about instance sizes. I noticed that the smallest available F2 instance is the 12xlarge, which includes two FPGAs. Are there plans to release smaller instances with a single FPGA, similar to the f1.2xlarge? If so, could you provide an estimated timeline for availability? 

Thanks.

kristopk, 2025-01-07:
Hi @itaibak, thanks for your patience and we are glad to hear you are ready to migrate to F2.  We expect to have the XDMA shell released very soon.  Once we have a release candidate (beta) branch, we will reply to this issue to let you know.  

Similar to F1, we plan to have a single FPGA instance size.  We will update you once available.

kristopk, 2025-02-11:
On Feb 5th, we released F2.6xlarge instances.  Delivering XDMA Shell for F2 instances is our highest priority for aws-fpga DevKit.

raghav-g13, 2025-03-17:
Hello, we also depend on XDMA shell availability for [FireSim](https://fires.im). Are there any updates on the timeline?

zwadood, 2025-05-27:
Hi @itaibak any ETA update for F2 XDMA shell?

ChrisKjellqvist, 2025-06-19:
Our tooling also relies on XDMA. Please provide an update on this.

suhwanhong999, 2025-07-01:
Asking for an update as well, it would be very nice to have Firesim running again.

marie-anne-xu, 2025-11-24:
Hi, are there any updates on the XDMA shell support?

ChrisKjellqvist, 2025-11-24:
@suhwanhong999 @marie-anne-xu Although XDMA and the usual AWS resources (APP_BAR4, etc...) don't seem to work for me, I found a workaround where I open `/dev/mem` as a file and the writes/reads on this file show up on the cl_sh_dma bus as they did for XDMA. I can provide more precise details if this is any use. I haven't verified the performance of this method, but it works.

rickydumplings, 2025-12-09:
Hi @ChrisKjellqvist, @marie-anne-xu and I are working on a similar method to replace XDMA functionality. Could you share more details on your approach?

ChrisKjellqvist, 2026-01-18:
@rickydumplings @prabhupraveen Sorry for the delay on this, I was trying to put together a more thorough explanation, but got side-tracked.

See here: https://github.com/Composer-Team/Beethoven-Software/blob/1ac4b448165e9db29c1c6827a0e6f51b07501c9d/runtime/src/data_server.cc#L128-L136

I found that the PCIE dma slave channel on the FPGA is still mapped into memory and is used for the SDA core. However, I had no success actually finding documentation of this or using it through normal means. But if you just mmap it into memory then, writes to this segment go over PCIE and show up on the bus as they would with XDMA. I can't remember which `lcpci` commands I used to find this address shown in the snippet above, but I was able to get repeatable results without the address changing. Hope this can be useful to someone.