aws/aws-fpga #673: AFI Creation with Vivado 2024

itaibak, 2025-01-07:
Hi, I am attempting to create an AFI after a successful build using the new Vivado 2024, but it fails to complete and returns the following error code when I run the aws ec2 describe-fpga-images command:

           "State": {
                "Code": "failed",
                "Message": "TOOL_VERSION_INVALID: The Tool Version provided is invalid. Invalid tool version 'v2024.1' (allowed versions: v2018.2, v2018.3, v2019.1, v2019.2, v2020.1, v2020.2, v2021.1, v2021.2, v2022.1, v2022.2, v2024.1)"
            },

It seems the new Vivado version should be supported, but it is still failing for some reason. I would appreciate any assistance with resolving this issue.
I am working on the branch `update_2024.1/f1_xdma_shell`
Thanks.

czfpga, 2025-01-07:
Thanks, @itaibak, we're still working on deploying to this region.  For now, if possible, please use `us-east-1` or `eu-west-2` for your test. We will keep you posted on this issue once the deployment is done.

coExpo, 2025-01-09:
this problem is same in region ap-northeast-2, please check this region too. 
thanks!

czfpga, 2025-01-21:
@itaibak @coExpo F1 AFI creation in the regions above have been enabled. Please test again.

itaibak, 2025-01-23:
It is working now, thanks for the support. I am closing the issue

closed 2025-01-23 by itaibak