aws/aws-fpga #675: Directory structure issue when generating .xclbin from .xo

ifransazov, 2025-01-15:
I am new to using HLS and Vitis tools so take this with a grain of salt. 

I generated a .xo file from the Vitis 2024.1 IDE and tried running the command: "**v++ -t hw --platform $AWS_PLATFORM -l <kernel name>.xo -o <kernel name>.xclbin**" 

after sourcing vitis_setup.sh and got this error: "**CRITICAL WARNING: [v++ 74-66] The hw platform file '/home/ubuntu/aws-fpga/vitis/aws_platform/xilinx_aws-vu47p-f2_202410_1/hw/hw.xsa' is not valid**"

After some ChatGPT it found that the expected directory structure in ~/aws-fpga/vitis/aws_platform/xilinx_aws-vu47p-f2_202410_1/ was supposed to look like: 

/home/ubuntu/aws-fpga/vitis/aws_platform/xilinx_aws-vu47p-f2_202410_1/
├── hw/
│   ├── hw.xsa
│   ├── hw.xsa.sha256
├── hw_emu/
│   ├── hw_emu.xsa
│   ├── hw_emu.xsa.sha256
├── sw/
│   ├── sw.spfm
│   ├── sw.spfm.sha256
├── xilinx_aws-vu47p-f2_202410_1.xpfm

but there were no hw/ hw_emu/ or sw/ subdirectories. I added them and moved the files in correctly and fixed the issue (as far as I can tell, I am still generating the binary). Let me know if you need more information or if this is not an error and I made a mistake with my v++ linking command above!

ifransazov, 2025-01-15:
I should also mention that the structure of the xilinx_aws-vu47p-f2_202410_1/ folder was what I had git cloned a few days ago so it appears to be the default currently. Letting you all know about this for the future if the structure creates issues

mjthimm, 2025-01-15:
Hi Ivan,

The directory structure that you've shown above is also under `/opt/Xilinx/Vitis/2024.1/platforms`. This is where `v++` should be looking for these files.

Could you share with me the location from which you were running the command to generate the `.xclbin` file above?

Below is a snippet of the `v++` commands used to build and link the `.xo` and `.xclbin` files for the `hello_world` Vitis_Accel_Examples directory.

```
############################## Setting Rules for Binary Containers (Building Kernels) ##############################
$(TEMP_DIR)/vadd.xo: src/vadd.cpp
        mkdir -p $(TEMP_DIR)
        v++ -c $(VPP_FLAGS) -t $(TARGET) --platform $(PLATFORM) -k vadd --temp_dir $(TEMP_DIR)  -I'$(<D)' -o'$@' '$<'

$(BUILD_DIR)/vadd.xclbin: $(TEMP_DIR)/vadd.xo
        mkdir -p $(BUILD_DIR)
        v++ -l $(VPP_FLAGS) $(VPP_LDFLAGS) -t $(TARGET) --platform $(PLATFORM) --temp_dir $(TEMP_DIR) -o'$(LINK_OUTPUT)' $(+)
        v++ -p $(LINK_OUTPUT) $(VPP_FLAGS) -t $(TARGET) --platform $(PLATFORM) --package.out_dir $(PACKAGE_OUT) -o $(BUILD_DIR)/vadd.xclbin
```

At a glance, your link command looks reasonable by comparison. 

I'll investigate why this is happening for your command and provide a response here.

mjthimm, 2025-01-16:
Hi Ivan,

What's happening here is that `v++` references the value of the environment variable `PLATFORM_REPO_PATHS` by default.

Please try redefining the environment variable `PLATFORM_REPO_PATHS` as follows:

`export PLATFORM_REPO_PATHS=/opt/Xilinx/Vitis/2024.1/platforms/xilinx_aws-vu47p-f2_202410_1.xpfm`

Once I did that, I was able to generate the `.xo` file by running `v++` from my home directory. But you could run it from anywhere you'd like:

```
ubuntu@ip-aaa-bb-cc-dd:~$ v++ -t hw --platform $PLATFORM_REPO_PATHS -k vadd -o $AWS_FPGA_REPO_DIR/vitis/examples/vitis_examples/hello_world/kern.xo $AWS_FPGA_REPO_DIR/vitis/examples/vitis_examples/hello_world/src/vadd.cpp
```

`-rw-rw-r--  1 ubuntu ubuntu 219079 Jan 16 17:48  kern.xo`

From there, you should be able to use the same tactic to link and create your `.xclbin`.

This change to `PLATFORM_REPO_PATHS` will be reflected in the release of the development kit where we enable Vitis AFI creation.

Best,
Max

ifransazov, 2025-01-17:
Thanks so much Max! Looking forward to the Vitis workflow being released!

Best,
Ivan

closed 2025-01-17 by ifransazov