aws/aws-fpga #676: create_vitis_afi.sh not in repo

ifransazov, 2025-01-15:
Hello aws team - I just generated an xclbin and was going to set the afi script to run and saw there is no current create_vitis_afi script in vitis/tools or anywhere else in the repo so far as I am aware - is there news on when the script will be updated from the f1 branch, or is there a new afi method? Thanks!

mjthimm, 2025-01-15:
Hi Ivan,

The creation of AFIs through Vitis isn't yet available on F2.

We're working hard on getting it ready and I will post a response here when it is released!

Until then, hardware emulation is available for the simulation of functional correctness of your designs.

Best,
Max

ifransazov, 2025-01-16:
Hi Max,

Thanks for letting me know, I appreciate it!

Best,
Ivan

closed 2025-01-17 by ifransazov