aws/aws-fpga #677: Could NOT find Vitis (missing: Vitis_AIE_INCLUDE_DIR)

coExpo, 2025-01-16:
I'm not sure if this is the right place to ask, but I tried running the hello_world example using the Vitis Unified IDE, and I encountered the following error:

 CMake Error at /opt/Xilinx/Vitis/2024.1/tps/lnx64/cmake-3.24.2/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
   Could NOT find Vitis (missing: Vitis_AIE_INCLUDE_DIR)
 Call Stack (most recent call first):
   /opt/Xilinx/Vitis/2024.1/tps/lnx64/cmake-3.24.2/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
   /opt/Xilinx/Vitis/2024.1/vitisng-server/scripts/cmake/FindVitis.cmake:133 (find_package_handle_standard_args)
   CMakeLists.txt:14 (find_package)
 
 
 -- Configuring incomplete, errors occurred!
 See also "/home/ubuntu/Documents/Example/hello_world/build/hw/CMakeFiles/CMakeOutput.log".
[ERROR] Build Failed

What should I do to resolve this issue?

i used FPGA Developer Ami(Ubuntu)

mjthimm, 2025-01-16:
Hello,

Once you cloned the development kit, did you source the `vitis_setup.sh` script from within the development kit?:

`. vitis_setup.sh`

If you did, did you also install the Xilinx Runtime (XRT)? The output from the setup includes a prompt that shows you the command that can be used to automatically install the XRT:

```
"###########################################################################"
"#                       No XRT install detected!                          #"
"#     To install the XRT for Vitis runtime support, run 'install_xrt'     #"
"###########################################################################"
```

Once you source `vitis_setup.sh`, this command, `install_xrt` becomes available and can be run at any time thereafter.

Vitis builds rely upon the XRT, so it must be installed before builds can be run.

Could you please show me the command you used to run the build that produced this error?

Best,
Max

coExpo, 2025-01-20:
i already source `vitis_setup.sh` and installed xrt(https://github.com/aws/aws-fpga/blob/f1_xdma_shell/Vitis/docs/XRT_installation_instructions.md)

and i used the vtis unified ide (i want to build on gui), so maybe there's no command.. i run the vitis gui and want to build on gui but that error occured.

mjthimm, 2025-01-21:
Could you let me know if you are using an F1 instance or an F2 instance?

If you're using an F2 instance, did you run `install_xrt` after sourcing `vitis_setup.sh`?

As for having a GUI to run Vitis through it's GUI, please see here:  https://github.com/aws/aws-fpga/blob/f2/developer_resources/DCV.md

This document contains instructions on how to start up a GUI for your F2 instance.