aws/aws-fpga #678: Access to modules over PCIS on small-shell

furkanturan, 2025-01-17:
Hello,

I found the examples quite explanatory, but still I am failing at communicating to my design over PCIS. Going over small-shell, I wish to access my hardawre over PCIS interface; high bandwidth transfers are not necessary for my design anyways. 

After failing to have it working on my full design, I want to test a very basic thing, connecting a BRAM to PCIS. Nothing else; no other interface is used, no HBM and no DDR. The design looks like this. The top verilog design is derived over CL_TEMPLATE, disabling all interfaces, except connecting PCIS to  BRAM.

```
+------------+      +-----------+      +--------------+      +------------+      +-----------+
| S_AXI_PCIS | <--> | AXI       | <--> | AXI          | <--> | AXI BRAM   | <--> | Block Mem |
|            |      | Reg Slice |      | SmartConnect |      | Controller |      | Generator |
+------------+      +-----------+      +--------------+      +------------+      +-----------+
```

I compiled it with `--mode small_shell` option, targetting the small shell.

```bash
./aws_build_dcp_from_cl.py -c test_design --mode small_shell --flow BuildAll
```

The AFI generation is successful, and I can load it into FPGA too.

My test application is as follows:

```C
pci_bar_handle_t pci_bar_handle = PCI_BAR_HANDLE_INIT;
int fpga_attach_flags = 0;

int pf_id  = 0;
int bar_id = 4;

rc = fpga_pci_attach(slot_id, pf_id, bar_id, fpga_attach_flags, &pci_bar_handle);
fail_on(rc, out, "Failed to attach to the AFI on slot id %d", slot_id);

uint64_t data_addr = 0x0000000000;

uint32_t wdata = 0xdeadbeef;
rc = fpga_pci_poke(pci_bar_handle, data_addr, wdata);
fail_on(rc, out, "Failed to write");

uint32_t rdata = 0;
rc = fpga_pci_peek(pci_bar_handle, data_addr, &rdata);
fail_on(rc, out, "Failed to read");

printf("wdata %x rdata %x\n", wdata, rdata);
```

For PCIS, I use pf_id 0 and bar_id 4. And, I expect, to read (peek) 0xdeadbeed as I written (poke) before. 

However, I always read 0.

___

The address mappings are correct; BRAM is accessible to PCIS over address 0x0000000000.

There is no implementation errors, no timing issues, no critical warnings, such as latches, multi-driven nets etc. 

At this point, I am quite lost. No clue where the problem might be. I am not even sure, if I should search for it in hw or sw.

I would appreciate any suggestions, if you can come up with any.

czfpga, 2025-01-17:
Hi @furkanturan,

The data path appears correct. Have you verified this design in simulation to check for issues like incorrect signal connections? I highly recommend simulation testing if you haven't done it yet.

To help isolate HW and SW issues, try directly accessing PF0 BAR4 using tools like `devmem2` to verify the register values. Also, please check and ensure the target card's BusMaster is enabled.

furkanturan, 2025-01-20:
Hello @czfpga 

Thanks a million for quick reply. You helped me solve it.

First, indeed busmaster was not enabled. I could not even guess it, as I heard it first time when reading your reply. 

Second, and more importantly, I found a bug in my design, such that two AXI signals names were wrong

I had this
```Veriog
.S_AXI_PCIS_wvalid    (cl_sh_dma_pcis_wvalid  ), // output              S_AXI_PCIS_wvalid
.S_AXI_PCIS_wready    (sh_ch_dma_pcis_wready  ), // input  wire         S_AXI_PCIS_wready
```
rather than
```Veriog
.S_AXI_PCIS_wvalid    (sh_cl_dma_pcis_wvalid  ), // output              S_AXI_PCIS_wvalid
.S_AXI_PCIS_wready    (cl_sh_dma_pcis_wready  ), // input  wire         S_AXI_PCIS_wready
```

The `cl_sh` and `sh_cl` parts of the wires names were swapped my mistake. As always, Vivado did not even create a meaning warning/error to let me spot the mistake. I found it through the protocol checker of simulation.

___

I had coded AXI many times in years, I can write it starting from a scratch document, but still these kind of typos happen. Though I like AXI, I got goosebumps when I need to manage AXI blocks in Verilog. SystemVerilog is better with interfaces, but my preference is block-designs in Vivado. I can connect AXI ports with one line of TCL, as in SystemVerilog; but on top of it, I can visualize my connections by opening the block-design in Vivado IPI.

Maybe that could be one feedback I can provide to this repo. When trying to simulate my design, following your suggestion, it was quite difficult, as I use TCL as part of my design. My source files are not simply `$(CL_DIR)/design/*.sv`. I could not find a nice way to incorporate my additional sources. I could do it for only a part of it, and caught the above-mentioned typo through the protocol checker. But it would have been nicer to be able put the whole design under simulation.

closed 2025-01-20 by furkanturan