aws/aws-fpga #681: Most Test Cases Failing on FPGA Developer AMI (Amazon Linux 2) – Peek_Poke Test Log Attached

gino-comp, 2025-02-06:
Hi,

I'm experiencing issues running the test cases from the aws_fpga repository on the FPGA Developer AMI (Amazon Linux 2). Most of the test cases are failing, and even the seemingly simplest one—the peek_poke test—is not working as expected.

Details:

Environment: FPGA Developer AMI (Amazon Linux 2)
Source Code: I have not made any modifications to the source code.
Test Case: Peek_Poke test (attached log file)
Observed Issue: The log indicates data mismatches and errors during the DDR memory verification phase (e.g., "Data Mismatch Addr" and "DDR Memory was overwritten").
I am looking for a working baseline that I can refer to so I can run simulations on my RTL and eventually generate an AFI. Any guidance on resolving these issues or adjustments needed in the testbench or environment settings would be greatly appreciated.

Thank you!

[test_ddr_peek_poke_212438.backup.log](https://github.com/user-attachments/files/18686743/test_ddr_peek_poke_212438.backup.log)

rcko-aws, 2025-02-06:
Looking at your attached log, I see the failure signature:  reading back "baad0000" .  
Yet I only see that hex signature in the cl_sde in sub example.  Are there any alterations from fresh clone?

I'll try replicating from fresh repo on ALMA

rcko-aws, 2025-02-06:
Try a clean clone first.  This is what I'm seeing:

b.card.u_ddr4_rdimm.rcd_enabled.NOT_LRDIMM.u_ddr4_dimm.rank_instances[1].mc_ca_mirroring_odd_rank.u_ddr4_rank.Micron_model.instance_of_sdram_devices[17].micron_mem_model.u_ddr4_model.always_diff_ck.if_diff_ck:Initialization complete @29578.515 ns
[   33692.000 ns] Finished DDR Initialization
[   33692.000 ns] Verifying DDR DIMM and Offsets
[   33692.000 ns] DDR DIMM Verification: Level 1 - Level 2
[   35512.000 ns]: Data Matched Addr: 400000000. Actual Data:aabbffcc00 <==> Expected Data: aabbffcc00

gino-comp, 2025-02-07:
Thank you for the suggestion to try a fresh clone of the repository. I hadn't changed any code, but it turns out I might have misconfigured some environment variables. After starting from a clean clone, the tests are now running as expected. I really appreciate the help and guidance!

closed 2025-02-07 by gino-comp