aws/aws-fpga #682: ddr_core.xci is no longer supported by 2020.2

dbwood298, 2025-02-11:
The current command hdk_setup.sh does no complete giving the following error

[image]

It looks like the ddr4_core.xci file is in a significantly different format than before.  
Commit comment statement says 
 
commit afdcf3f8d683bd5fa1e8b3bb3ac3a85deb5fe3b5
Author: Chen Zhu <zhcz@amazon.com>
Date:   Tue Dec 17 22:08:03 2024 -0600
 
    Update all IPs to Vivado 2024.1

My guess is that the change to the xci is not compatible with 2020.2
the git hub page still says
⚠️ NOTE: CL's with Small Shell F1.S.1.0 require Xilinx 2024.1 or 2020.2 tools

czfpga, 2025-02-11:
Hi @dbwood298,

For the latest [F1.S.10 Small Shell](https://github.com/aws/aws-fpga/tree/f1_small_shell?tab=readme-ov-file), please use the new [Ubuntu devAMI ](https://aws.amazon.com/marketplace/pp/prodview-f5kjsenkfkz5u) with 2024.1 Vivado.

dbwood298, 2025-02-11:
Our goal is to use Vivado 2020.2 with builds locally.   If you are saying 2020.2 is no longer supported this needs to be stated

czfpga, 2025-02-11:
You can still build with Vivado 2020.2 locally using release 1.6.0. Release 1.6.1 is for supporting the new DevAMI listed [here](https://github.com/aws/aws-fpga/tree/f1_small_shell?tab=readme-ov-file#fpga-developer-ami).

dbwood298, 2025-02-11:
Thanks for your help.  I'm sorry for this last question.   How do you checkout release 1.6.0 of the aws-fpga f1-small-shell repo.  
There doesn't seem to be a tag for it, or a release for it.  Is it a specific commit?

czfpga, 2025-02-11:
For 1.6.0, please use [commit 73dd58a](https://github.com/aws/aws-fpga/commit/73dd58a40d783d97dddb2abb0913ebdc3dfcf5db). That's definitely a miss on our end. We'll fix it shortly. Thanks for pointing it out.

czfpga, 2025-02-11:
Tags have been applied to the F1 branches. Here is the one for [v1.6.0 small shell](https://github.com/aws/aws-fpga/tree/v1.6.0). Thanks

closed 2025-04-21 by TrahStevAmz