aws/aws-fpga #685: Unable to setup a custom CL on FPGA and run it through my host script

PratMaha, 2025-02-19:
I am trying to send data through the PCIE interface to my custom logic, but I'm not able to get anything to work. I would appreciate some support on the same since the documentation you have provided has not been helpful.

czfpga, 2025-02-19:
@PratMaha,

Have you fully verified the CL in simulation to ensure the it works as expected? Please refer to [this doc ](https://github.com/aws/aws-fpga/blob/f2/hdk/docs/RTL_Simulation_Guide_for_HDK_Design_Flow.md)for details of running RTL simulation.  If the design has passed the simulations, I will recommend check and ensure the correct AFI is loaded to the instance. Before passing large chunk of data, start with something basic, like accessing a CL register to ensure the shell interfaces are functional.

PratMaha, 2025-02-19:
I have verified my design on Vivado locally but I'm getting this error on AWS : 
ERROR: [VRFC 10-3006] '/home/ubuntu/src/project_data/aws-fpga/hdk/common/verif/ip_simulation_libraries/xsim_compiled_cl_ips/cl_gpu_base_test.sdb' needs to be re-saved since 'xil_defaultlib.$unit_assert_sv_2159410909' changed
ERROR: [VRFC 10-3032] 'xil_defaultlib.cl_gpu_base_test' failed to restore
ERROR: [XSIM 43-3225] Cannot find design unit xil_defaultlib.cl_gpu_base_test in library work located at xsim.dir/work.
make: *** [/home/ubuntu/src/project_data/aws-fpga/hdk/common/verif/tb/scripts/Makefile.xsim.inc:52: compile] Error 1

czfpga, 2025-02-19:
Refer to one of HDK examples for CL top module I/O connection: https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_sde/design/cl_sde.sv#L23-L27. The CL-Shell interfaces are already defined and fixed [here](https://github.com/aws/aws-fpga/blob/f2/hdk/common/shell_stable/design/interfaces/cl_ports.vh), they cannot be modified.

PratMaha, 2025-02-19:
So, to clarify, these are the interfaces I use to connect my host to my CL design right?

czfpga, 2025-02-19:
Correct, the Shell and CL are connected through these interfaces. 

For the host code, you can refer to the [runtime example code](https://github.com/aws/aws-fpga/blob/f2/hdk/README.md#software) provided in the CL examples.

czfpga, 2025-02-19:
I strongly recommend reviewing the document above and going through all the CL examples to get familiar with the HDK development environment first. That will help you quickly find out the files needed in the repo as well as build your own example. The code examples can be found in the `software/runtime/` directory of each example.

PratMaha, 2025-02-19:
There are no examples here that have runtime code for SH to CL communication. Can you point me to one? All that's there is a null example

PratMaha, 2025-02-19:
Also, how do I resolve such synthesis errors :
RROR: [Synth 8-5809] Error generated from encrypted envelope. [/home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/src_post_encryption/global_mem_controller_synth.sv:35]
INFO: [Synth 8-9084] Verilog file '/home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/src_post_encryption/global_mem_controller_synth.sv' ignored due to errors
ERROR: [Synth 8-5809] Error generated from encrypted envelope. [/home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/src_post_encryption/gpu_die.sv:54]
ERROR: [Synth 8-5809] Error generated from encrypted envelope. [/home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/src_post_encryption/gpu_die.sv:2]
ERROR: [Synth 8-5809] Error generated from encrypted envelope. [/home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/src_post_encryption/cl_gpu.sv:2]
Like I said, I've run everything on Vivado locally, but it's failing on AWS

czfpga, 2025-02-19:
You can comment out these [two lines](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_sde/build/scripts/encrypt.tcl#L77-L82) in your example directory. It will stop the source encryption so that you can see the true errors.

PratMaha, 2025-02-24:
Is there any basic example that actually completely works (and doesn't have caveats like XDMA not supported etc etc) for Host->FPGA PCIE communication?
I want to minimise the effort involved in moving from my local Vivado to the AWS (I don't have enough time to ramp up and learn every detail of your examples), but there's no useful example that just demonstrates basic Host<-> FPGA PCIE communication

PratMaha, 2025-02-26:
Okay, so I'm getting this error right now :
tb.card.u_ddr4_rdimm.rcd_enabled.NOT_LRDIMM.u_ddr4_dimm.rank_instances[1].mc_ca_mirroring_odd_rank.u_ddr4_rank.Micron_model.instance_of_sdram_devices[17].micron_mem_model.u_ddr4_model:Configured as x4 16G stack:1
    5052.000 ns : tb.card.fpga.sh.axi_pc_mstr_inst_pcim.REP : BIT(          0) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axi_pc_mstr_inst_pcim.REP : BIT(          7) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axi_pc_mstr_inst_pcim.REP : BIT(         37) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axi_pc_mstr_inst_pcim.REP : BIT(         44) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axi_pc_mstr_inst_pcim.REP : BIT(         90) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axi_pc_mstr_inst_pcim.REP : BIT(         91) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_ocl_slv_inst.REP : BIT(         59) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_ocl_slv_inst.REP : BIT(         60) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_ocl_slv_inst.REP : BIT(         83) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_ocl_slv_inst.REP : BIT(         84) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_sda_slv_inst.REP : BIT(         59) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_sda_slv_inst.REP : BIT(         60) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_sda_slv_inst.REP : BIT(         83) :   ERROR : Invalid state x
    5052.000 ns : tb.card.fpga.sh.axl_pc_sda_slv_inst.REP : BIT(         84) :   ERROR : Invalid state x
[    6100.500 ns] : Detected   0 errors
[    6100.500 ns] : Checking protocol checker error status...
[    6100.500 ns] : *** 'X' propagation detected in protocol checker status bits. Please dump waves and look at pc_status bits for more information***

This is my top-level module :
module cl_gpu(
    input clk_main_a0,
    input rst_main_n,

    // PCIS AXI-4 interface between host and cl
    // Address Write Signals
    input logic [15:0]  sh_cl_dma_pcis_awid,
    input logic [63:0]  sh_cl_dma_pcis_awaddr,
    input logic [7:0]   sh_cl_dma_pcis_awlen,
    input logic [2:0]   sh_cl_dma_pcis_awsize,
    input logic [1:0]   sh_cl_dma_pcis_awburst,
    input logic         sh_cl_dma_pcis_awvalid,
    output logic        cl_sh_dma_pcis_awready,
    // Data Write Signals
    input logic [15:0]  sh_cl_dma_pcis_wid,
    input logic [511:0] sh_cl_dma_pcis_wdata,
    input logic [63:0]  sh_cl_dma_pcis_wstrb,
    input logic         sh_cl_dma_pcis_wlast,
    input logic         sh_cl_dma_pcis_wvalid,
    output logic        cl_sh_dma_pcis_wready,
    // Response Write Signals
    output logic [15:0] cl_sh_dma_pcis_bid,
    output logic [1:0]  cl_sh_dma_pcis_bresp,
    output logic        cl_sh_dma_pcis_bvalid,
    input logic         sh_cl_dma_pcis_bready,
    // Address Read Signals
    input logic [15:0]  sh_cl_dma_pcis_arid,
    input logic [63:0]  sh_cl_dma_pcis_araddr,
    input logic [7:0]   sh_cl_dma_pcis_arlen,
    input logic [2:0]   sh_cl_dma_pcis_arsize,
    input logic [1:0]   sh_cl_dma_pcis_arburst,
    input logic         sh_cl_dma_pcis_arvalid,
    output logic        cl_sh_dma_pcis_arready,
    // Data Read Signals
    output logic [15:0] cl_sh_dma_pcis_rid,
    output logic [511:0] cl_sh_dma_pcis_rdata,
    output logic [1:0]  cl_sh_dma_pcis_rresp,
    output logic        cl_sh_dma_pcis_rlast,
    output logic        cl_sh_dma_pcis_rvalid,
    input logic         sh_cl_dma_pcis_rready
);

And this is my testbench :
`include "common_base_test.svh"

module cl_gpu_base_test();
   import tb_type_defines_pkg::*;

   initial begin
      tb.power_up();
      #500
      tb.power_down();
      report_pass_fail_status();

      $finish;
   end
endmodule

What's going on? Why is such a simple testbench not working?

PratMaha, 2025-02-26:
**So I'm also getting this error when I try to push my design through the RTL->AFI flow :**
AWS FPGA: (22:27:44): Start linking customer design cl_gpu

## add_files ${AWS_DCP_DIR}/cl_bb_routed.${SHELL_MODE}.dcp
## add_files ${checkpoints_dir}/${CL}.${TAG}.post_synth.dcp
## set_property SCOPED_TO_CELLS {WRAPPER/CL} \
##              [get_files ${checkpoints_dir}/${CL}.${TAG}.post_synth.dcp]
## link_design -mode default \
##             -reconfig_partitions {WRAPPER/CL} \
##             -top top
Command: link_design -mode default -reconfig_partitions WRAPPER/CL -top top
Design is defaulting to srcset: sources_1
Design is defaulting to constrset: constrs_1
Design is defaulting to dcp part: xcvu47p-fsvh2892-2-e
INFO: [Project 1-454] Reading design checkpoint '/home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/checkpoints/cl_gpu.2025_02_26-222645.post_synth.dcp' for cell 'WRAPPER/CL'
1 Infos, 0 Warnings, 0 Critical Warnings and 1 Errors encountered.
link_design failed
ERROR: [Netlist 29-77] Could not replace (cell 'cl_simple_bb', library 'work_CL_CL_1', file 'NOFILE') with (cell 'cl_gpu', library 'work', file 'cl_gpu.edf') because of a port interface mismatch; 3235 ports are missing on the replacing cell. 5 of the missing ports are: 'CLK_DIMM_DN' 'CLK_DIMM_DP' 'M_ACT_N' 'sh_cl_status_vdip[1]' 'sh_cl_status_vdip[0]'.
Resolution: Modify RTL to reference correct ports from the netlist

    while executing
"source ${scripts_dir}/build_level_1_cl.tcl"
    ("default" arm line 3)
    invoked from within
"switch $BUILD_FLOW {
  "SynthCL" {
    source ${scripts_dir}/synth_${CL}.tcl
  }

  "ImplCL" {
    source ${scripts_dir}/build_level_1_cl.tcl
  }

  d..."
    (file "build_all.tcl" line 265)
INFO: [Common 17-206] Exiting Vivado at Wed Feb 26 22:27:46 2025...
Save the existing to_aws/ to to_aws_backup_2025_02_26-222747/
ERROR: Did not find the post-route DCP file from /home/ubuntu/src/project_data/aws-fpga/hdk/cl/examples/cl_gpu/build/checkpoints/

Any idea what I could be doing wrong?

PratMaha, 2025-03-06:
So, I've been able to make some progress but the peeks and pokes into my CL are not working. I tried opening the waveform to debug using the steps given in the RTL simulation guide but that just gave me an empty waveform…., so I'm not even sure if things are getting set by the PCIE host. This is the group of lines running the peek and poke operations :
   task process_instructions();
      int file;         // file handle
      string line;      // each line from file
      int status;       // fscanf/gets status
      reg [31:0] value; // holds parsed data from each line
      int i;

      file = $fopen("/home/ubuntu/aws-fpga/hdk/cl/examples/cl_gpu/verif/tests/gpu_instructions.txt", "r");
      if (!file) begin
         $display("[%0t] ERROR: Could not open gpu_instructions.txt", $time);
         return;
      end
      
      i = 0;
      while (!$feof(file)) begin
         status = $fgets(line, file);
         if (status) begin
            // Ignore commented lines
            if (line[0] == "#") begin
               continue;
            end

            // Check the first letter for S/D/R
            case (line[0])
               "S": begin
                  value = parse_data(line, 0);  // 0 => STATE
		  $display("Line : ", line);
		  $display("Value : ", value);
                  // cpu_instr_type=0, cpu_in_data=value => encode in 64 bits
                  // [32] = instr_type, [31:0] = data
                  tb.poke( 
                     0, // address = 0, because all are "0 addresses"
                     64'h0,
                     {31'b0, 1'b0, value}, // 0 for STATE
                     0, DataSize::UINT64
                  );
		  $display("Data sent");
                  // Wait for GPU ack (optional). For example, if your hardware
                  // sets an ack bit in the same 64'h0 location, you can poll:
                  // wait_for_ack(1); // pass a bit or offset if needed
		  $display("Ack received");
               end

               "D": begin
                  value = parse_data(line, 1);  // 1 => DATA
                  // cpu_instr_type=1, cpu_in_data=value => encode in 64 bits
                  tb.poke(
	             0,
                     64'h0,
                     {31'b0, 1'b1, value}, // 1 for DATA
                     0, DataSize::UINT64
                  );
                  // wait_for_ack(1);
               end

               "R": begin
                  // READ_DATA => also sets cpu_instr_type=1, but data=0
		  tb.peek(0, 0, value);
                  $display("Output[%0d] = %h", i, value[31:0]);
                  // wait_for_ack(1);
		  i = i + 1;
               end

               default: begin
                  // Possibly handle unknown lines
                  $display("[%0t] WARNING: Unknown instruction '%s'", $time, line);
               end
            endcase

         end
      end
      $fclose(file);
   endtask

   //--------------------------------------------------------------------------
   // 3) Function to parse data from a line (for STATE or DATA)
   //--------------------------------------------------------------------------
   function [31:0] parse_data(input string line, input bit is_data);
      reg [31:0] local_val;
      int parse_ok;
      if (is_data == 1) begin
         // "DATA 0x..."
         parse_ok = $sscanf(line, "DATA %h", local_val);
      end else begin
         // "STATE 0x..."
         parse_ok = $sscanf(line, "STATE %h", local_val);
	 $display("Local val :", local_val);
      end
      if (parse_ok == 0) begin
         $display("[%0t] WARNING: Failed to parse line: '%s'", $time, line);
         local_val = 32'h0;
      end
      return local_val;
   endfunction

I can see line and value are getting set correctly, how can I debug this further? I've run the simulation for 20000 us, which should be more than enough for a single poke to happen?

czfpga, 2025-03-07:
Which simulator are you using? Does the simulator spit any any critical warning or error? 

Can you also try passing the function arguments using name so it's more clearer to see which interface you're using to access CL?

PratMaha, 2025-03-07:
I'm using the standard simulator, which is xsim I believe?
Nope, there's no warning or error, it's just…. hanging
>>Can you also try passing the function arguments using name so it's more clearer to see which interface you're using to access CL?
What do you mean by this? I'm simply using cl_ports.vh as the top-level module arguments? Also, I'm using PCIS, but I'm not sure if it's configured correctly

czfpga, 2025-03-07:
Without looking directly at your setup, it's very difficult to pinpoint where exactly the problem is. 

I suggest you port over your CL logic to one of our examples, e.g. CL_DRAM_HBM_DMA. Keep the original CL name but replace the CL logic body with your own design. And modify one of its existing tests, e.g. [test_peek_poke](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/hdk/cl/examples/cl_dram_hbm_dma/verif/tests/test_peek_poke.sv#L30-L34), by replacing highlighted test body with your own test code above. 

I think this would help minimize the chance of having setup issues. Try this first and see if the simulation runs through .

czfpga, 2025-03-07:
I also recommend checking out the AWS [rePost](https://repost.aws/) forum, especially the posts with the `FPGA Development` tag. You might find similar questions that have been previously answered and additional community support there.

ai-hpc, 2025-06-03:
Thanks I resolved issue.