aws/aws-fpga #687: F2 questions

ljp0101, 2025-03-07:
1) Is there a target date for XDMA support? Would help for my planning.

2) I can't purchase reserved F2 instances on the console. Am I doing something wrong? Do I need to apply for special permissions like I think I may have done at some point for F1s?

3) Standard pattern is for releases to show up on github on the mid-right panel, which is stuck on v2.0.1.

TrahStevAmz, 2025-03-07:
Hello @ljp0101 

1. I can only share that we are working on it. I am unable to comment on the timeline.
2. What specific error are you getting when you attempt to purchase an F2 instance?
3. I appreciate the feedback. I applied a tag on the latest commit and created a new/latest release.

TrahStevAmz, 2025-03-10:
Because your remaining question is not related to the aws-fpga repository, please post this question on repost.aws. Be sure to add the FPGA Development tag on it so it will alert the appropriate teams.

TrahStevAmz, 2025-04-21:
Hello @ljp0101,

We haven't heard from you on repost. I'd recommend reaching out to your account manager to contact us about the problem.

closed 2025-04-21 by TrahStevAmz